RobotLocomotion/drake #4893: snopt solver thread safety?

jefesaurus, 2017-01-24:
Has anyone been able to use `SnoptSolver` on different optimization problems from multiple threads? I'm linking the F2C version of snopt which I added `f2c -a` so it would use regular local stack allocation instead of static, but I still get segfaults all over the place.

After adding some mutexes into `snopt_solver.cc` where it allocates the workspaces and sets problem parameters, I can lower the frequency of segfaults, but it isn't good enough.

I'm not intimately familiar with the workings of snopt so I'm just curious if this has been done before, or if I'm chasing an endless rabbit hole. Would linking the fortran version help?

RussTedrake, 2017-01-25:
I'm not surprised to hear that our current snopt_solver is not thread safe.  Unfortunately, I can't think of a recent example of running snopt in multiple threads.  @patmarion did you ever have multiple snopt threads running from director?

hongkai-dai, 2017-01-25:
I never tried that. But SNOPT manual says it can potentially run on multiple thread.http://web.stanford.edu/group/SOL/guides/sndoc7.pdf

sammy-tri, 2017-01-25:
I believe our implementation of SnoptSolver is intended to be thread safe, and I don't see any really obvious gotchas in the implementation there (but manual inspection of a single file is hardly a proof of thread-safe code).

You could mutex all of Solve(), but of course this might defeat the point of using threads.

Any chance you've got a backtrace of where the segfault is happening?  It might not help since it could be far away from the problem, but I'm curious.  Another option would be to run a single thread of your program in valgrind and see if we've done something stupid with the allocation code in SnoptSolver which is causing you to run off the end of a buffer (it's happened before in snopt_solver.cc).  Or maybe try to helgrind (or drd) (both valgrind tools) to see if they can shed any light on thread issues.  IIRC, helgrind is quite slow.

patmarion, 2017-01-25:
No I have not used snopt (or the drake IK interface) in a multi-threaded scenario.  I have used it in a multi-processing scenario however.  I looked at the PDF and indeed it says that the snopt routines may be used by multi-threaded applications.

jefesaurus, 2017-01-25:
I'll try to get a stack trace/valgrind in a bit, but it's not stable, and clearly something is smashing something else's data, so it's not super useful.

It really seems like it's close to working, which is why I'm suspicious of the f2c conversion... Once I turn `iPrint=-1`; and iSumm=-1` so the threads aren't trying to write to the same file, it resolves most of the initialization segfaults, and I can solve a few problems in parallel, but eventually it chokes.

Also, would it work to use c++ function binding rather than the snopt user workspace to pass the drake problem instance to the `user_fun`?

jwnimmer-tri, 2017-01-25:
For data races, the helgrind tool that Sam mentioned is great for making problem reports more deterministic, and for reporting the first indication of a problem, rather than some segfault down the road.

jwnimmer-tri, 2018-03-22:
Having just looked at all of the static globals in the f2c version, I'm not surprised it doesn't work.  Maybe #7984 is the best resolution here?

jefesaurus, 2018-03-23:
That's my thought too. I'd love to be able to solve NLPs in parallel without multiprocessing or using an old [patch](https://sourceforge.net/p/nuto/wiki/MUMPS%20Solver%20package/) applied to an old mumps rolled into an old ipopt which is the only other way I've discovered thus far.

jwnimmer-tri, 2018-10-25:
FYI Within TRI we have a draft of a fix for this -- by using the Fortran direct interface to SNOPT, it no longer has global variables.  We're actively now working to contribute this back to Drake master.

EricCousineau-TRI, 2018-12-10:
Relates #9633

jwnimmer-tri, 2019-01-03:
There is now a method `SnoptSolver::is_thread_safe` to report whether the build of SNOPT wrapped into Drake is thread-safe or not.  To enable the thread-safe build, use `--config snopt_fortran` instead of `--config snopt` when compiling via bazel (not yet exposed through a CMake knob).

I think what remains for this issue is to either offer the CMake option for Fortran (or perhaps the Fortran version becomes the everywhere default), and to make sure we have CI coverage for it.

jamiesnape, 2019-01-04:
> perhaps the Fortran version becomes the everywhere default

I would go for this and strongly consider removing the F2C version completely.

jwnimmer-tri, 2019-01-04:
I think that's exactly the right medium-term goal.  In the short term though, both 7.4/Fortan and 7.6/Fortran have (different) bugs in their implementation, so I'm hesitant to force it on people too quickly.  I think once those bugs are fixed, we should quickly make it the default.

jwnimmer-tri, 2019-01-16:
FYI I'm consolidating the work blocking such a transition in #10422.

jwnimmer-tri, 2019-01-25:
> ... by using the Fortran direct interface to SNOPT, it no longer has global variables.

It turns out that this isn't entirely true.  It's true that by using the Fortran interface, we no longer have f2c adding globals everywhere.  However, `snopt_wrapper.f90` declares the character workspace `cw` as a global variable, and early indications are that it over it all the time, during sninit, snmem, snker, etc.

jwnimmer-tri, 2019-01-30:
Now that PR #10522 is merged, this problem is resolved iff we're using the Fortran version.  Thus, I'll close this as a duplicate of #10422 (i.e., making Fortran the default).

closed 2019-01-30 by jwnimmer-tri