RobotLocomotion/drake #4910: Remove functions that add cost/constraints in MathematicalProgram for all variables.

hongkai-dai, 2017-01-25:
Currently `MathematicalProgram` has functions like
```
MathematicalProgram::AddLinearConstraint(A, lb, ub);
```
which means imposing the constraint `lb <= A*x<=ub` for ALL variables inside `MathematicalProgram`. Personally I find this function unnecessary. This function also assumes that the user knows the order of the variables stored inside `MathematicalProgram`, as by reversing the order inside `x`, the meaning of this constraint is totally changed. I think that is a very strong assumption, and the user in most cases do not know the order of the variables inside `MathematicalProgram`.

Also, removing this function makes `mathematical_program.h` file a lot smaller, since for each type of cost/constraint, we have a function for all variables.

@sammy-tri and @RussTedrake what do you think?

RussTedrake, 2017-01-25:
I completely agree.  Let's zap them

rdeits, 2017-01-25:
I also totally agree.

sammy-tri, 2017-01-26:
Fine by me.

PR #4943 (2017-01-31): Remove add cost for all variables
Resolves #4910 

@RussTedrake you might like the changes in `convex_optimization_test.cc`, when writing linear equality constraint for matrices. Thanks to @soonho-tri 's work on symbolic expression, this part is a lot cleaner.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4943)
closed 2017-01-31 by hongkai-dai