RobotLocomotion/drake #4911: Using Eigen::Array<Polynomial, Dynamic, Dynamic> for PolynomialMatrix in PiecewisePolynomial

hongkai-dai, 2017-01-25:
Currently we use `Eigen::Matrix<Polynomial, Eigen::Dynamic, Eigen::Dynamic>` as the type of `PolynomialMatrix` inside `PiecewisePolynomial` class. But this can cause confusion since we are doing element-wise product, when multiplying two piecewise polynomials, as we have seen in #4854 

I think the solution is to use `Eigen::Array` instead of `Eigen::Matrix`, as
```C++
typedef Eigen::Array<PolynomialType, Eigen::Dynamic, Eigen::Dynamic> PolynomialMatrix;
```
`Eigen::Array` supports element-wise product and addition.
@siyuanfeng-tri what do you think?

hongkai-dai, 2017-01-25:
Also @jefesaurus

siyuanfeng-tri, 2017-01-26:
@hongkai-dai, that should be the correct representation. I will fix it at some point. 
cc @tkoolen @rdeits @RussTedrake, just to make sure nothing is relying on the current "matrix multiplication" behavior.

RussTedrake, 2017-01-26:
no dependencies from anything I've written.

rdeits, 2017-01-26:
I don't think I've relied on that behavior anywhere.

tkoolen, 2017-01-26:
I can't think of anything either.

jwnimmer-tri, 2019-03-04:
@siyuanfeng-tri Want to take a crack at solving this?  Or else re-assign?

siyuanfeng-tri, 2019-03-27:
I am tempted to just close it due to lack of strong desire and relative low value added to drake. Moving to backlog for now.

jwnimmer-tri, 2023-01-18:
Given https://github.com/RobotLocomotion/drake/pull/4854#issuecomment-614764856, this cleanup request is mathematically wrong anyway.

closed 2023-01-18 by jwnimmer-tri