RobotLocomotion/drake #4914: Mac builds with Matlab failing in compile of Iris

ggould-tri, 2017-01-26:
Starting two nights ago, but masked by other build failures yesterday, all mac builds with matlab began failing during the compilation of the `iris` external.

Example failures: https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=331321
Relevant commits listed here: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-clang-nightly-matlab-debug/172/

Typical errors are:
...`iris_wrapperPYTHON_wrap.cxx:4793:12: error: explicit specialization of non-template struct 'traits_reserve'`
`cddlp.c:2954:109: warning: data argument not used by format string [-Wformat-extra-args]
    if (localdebug) printf("No interior-point is found and thus the standard LP technique will be used.\n", ired);`
`cddio.c:1948:16: warning: assigning to 'char *' from 'const char *' discards qualifiers [-Wincompatible-pointer-types-discards-qualifiers]`
`cddlib.c:116:59: warning: for loop has empty body [-Wempty-body]
      for (itemp=1; cone->OrderVector[itemp]!=hh; itemp++);`
`cddlp.c:2954:109: warning: data argument not used by format string [-Wformat-extra-args]
    if (localdebug) printf("No interior-point is found and thus the standard LP technique will be used.\n", ired);`
`cddio.c:1948:16: warning: assigning to 'char *' from 'const char *' discards qualifiers [-Wincompatible-pointer-types-discards-qualifiers]`
`cddlib.c:116:59: warning: for loop has empty body [-Wempty-body]
      for (itemp=1; cone->OrderVector[itemp]!=hh; itemp++);`
`cddlp_f.c:2955:109: warning: data argument not used by format string [-Wformat-extra-args]
    if (localdebug) printf("No interior-point is found and thus the standard LP technique will be used.\n", ired);`
`cddio_f.c:1949:16: warning: assigning to 'char *' from 'const char *' discards qualifiers [-Wincompatible-pointer-types-discards-qualifiers]`
`cddlib_f.c:117:59: warning: for loop has empty body [-Wempty-body]
      for (itemp=1; cone->OrderVector[itemp]!=hh; itemp++);`

rdeits, 2017-01-26:
This is a result of our patched version of `pycontainer.swg`, which is used by SWIG to wrap `std::vector` and the like. The version in SWIG has a bug when used with types like Eigen's, so we ship a patched version in `swigmake`. Unfortunately, that makes us very sensitive to the SWIG version. Specifically, I know that SWIG 3.0.11 is incompatible with that patch, hence https://github.com/rdeits/swigmake/pull/6

Now that we're moving away from drake shipping its own copy of SWIG, we can't rely on an exact version, so this patched `pycontainer.swg` approach won't work. I'm fine with disabling IRIS in the drake superbuild until I can resolve the situation (possibly by switching IRIS from SWIG to pybind11).

rdeits, 2017-01-26:
This will be fixed when https://github.com/rdeits/iris-distro/pull/52 comes in (since it removes the entire SWIG layer from IRIS)

PR #4944 (2017-01-30): update IRIS version to remove swig, mex, and pods dependencies
This updates to the latest IRIS master, which uses pybind11 instead of SWIG and which does not build any Matlab mex files or use any of the Pods architecture. 

This *should* preserve IRIS compatibility with Drake, since the Matlab functions still work (they just no longer have mex-ed speedups). However, there's one significant change: Previously, IRIS (like all our other Matlab tools) installed an `addpath_iris.m` file which drake's `checkDependency` would call to mutate the global path and add the IRIS folders. Now, instead, IRIS [... cut]
closed 2017-01-30 by jamiesnape