RobotLocomotion/drake #4916: VTK Build Warning Indicating Use of Deprecated Header

liangfok, 2017-01-26:
# The Warning

```
In file included from /usr/include/c++/4.9/backward/strstream:51:0,
                 from /usr/include/vtk-5.8/vtkIOStream.h:112,
                 from /usr/include/vtk-5.8/vtkSystemIncludes.h:40,
                 from /usr/include/vtk-5.8/vtkIndent.h:24,
                 from /usr/include/vtk-5.8/vtkObjectBase.h:43,
                 from /usr/include/vtk-5.8/vtkObject.h:41,
                 from /usr/include/vtk-5.8/vtkViewport.h:32,
                 from /usr/include/vtk-5.8/vtkRenderer.h:32,
                 from /home/ubuntu/workspace/linux-gcc-ninja-experimental/drake/systems/sensors/rgbd_camera.cc:3:
/usr/include/c++/4.9/backward/backward_warning.h:32:2: warning: #warning This file includes at least one deprecated or antiquated header which may be removed without further notice at a future date. Please use a non-deprecated interface with equivalent functionality instead. For a listing of replacement headers and interfaces, consult the file backward_warning.h. To disable this warning use -Wno-deprecated. [-Wcpp]
 #warning \
  ^
```

# Example Log

* https://drake-cdash.csail.mit.edu/viewBuildError.php?type=1&buildid=334384, which is part of https://drake-jenkins.csail.mit.edu/job/linux-gcc-ninja-experimental/2824/.

liangfok, 2017-01-26:
Maybe this is because we're using an old version of VTK? Of so, feel free to close this issue without resolution.

kunimatsu-tri, 2017-01-26:
For the record, the warning is coming from vtkIOStream.h where deprecated std::strstream is included.

If we update VTK to newer version than 6.0, this problem will be gone.

closed 2017-01-26 by kunimatsu-tri