RobotLocomotion/drake #4929: Mysterious Bazel CI Error: No toolchain found

liangfok, 2017-01-27:
# The Error

```
18:02:07 ERROR: No toolchain found for --cpu='k8' --compiler='gcc-4.9'. Valid toolchains are: [
18:02:07   --cpu='armeabi-v7a' --compiler='compiler' --glibc='armeabi-v7a',
18:02:07   --cpu='k8' --compiler='compiler' --glibc='local',
18:02:07   --cpu='x64_windows_msvc' --compiler='cl' --glibc='msvcrt140',
18:02:07 ].
18:02:07 ____Elapsed time: 4.599s
18:02:07 ERROR: Couldn't start the build. Unable to run tests.
```

# Example Logs

* https://drake-jenkins.csail.mit.edu/job/linux-gcc-bazel-experimental/839/

david-german-tri, 2017-01-27:
Note the first line of the console log:

> GitHub pull request #4922 of commit 5497265c66beab1da7da4c43ffef9a4c6666a8eb, has merge conflicts.

Here's my guess: The PR being tested had merge conflicts with the HEAD of master.  Therefore, Jenkins tested the branch without merging, and the branch had not yet pulled in #4772.  As a result, the new explicit Bazel toolchains that the CI script attempts to use were not present, which would produce that error.

I'm going to close this.  Feel free to reopen if it recurs.

closed 2017-01-27 by david-german-tri

liangfok, 2017-01-27:
Perhaps abort the build if there is a merge conflict?

david-german-tri, 2017-01-27:
> Perhaps abort the build if there is a merge conflict?

No, I don't really think that's what most people would or should want.

jwnimmer-tri, 2017-01-27:
(Sometimes people want to run their branch on CI prior to working through the merge.  The hazard of CI and unmerged master being incompatible is low.)

david-german-tri, 2017-01-27:
> The hazard of CI and unmerged master being incompatible is low.

Especially since there is a separate interlock on merge conflicts.