RobotLocomotion/drake #4931: Python link issues with pybind11 inside Drake

rdeits, 2017-01-27:
I was running a quick test to see if I could swap the RigidBodyTree and IK bindings from SWIG to pybind11, but I'm running into some linking issues: 

```
[1/1] Linking CXX shared module pybind/rbtree
FAILED: pybind/rbtree
: && /usr/local/opt/ccache/libexec/c++  -Werror=all -Werror=ignored-qualifiers -Werror=overloaded-virtual -Werror=shadow -Werror=inconsistent-missing-override -Werror=sign-compare -Werror=return-stack-address -g  -bundle -Wl,-headerpad_max_install_names -Wl,-undefined -Wl,error -o pybind/rbtree pybind/CMakeFiles/rbtree.dir/rbtree.cc.o  -undefined dynamic_lookup multibody/parsers/libdrakeMultibodyParsers.dylib multibody/libdrakeRBM.dylib multibody/collision/libdrakeCollision.dylib /Users/rdeits/locomotion/drake-distro/build/install/lib/libBulletDynamics.dylib /Users/rdeits/locomotion/drake-distro/build/install/lib/libBulletCollision.dylib /Users/rdeits/locomotion/drake-distro/build/install/lib/libLinearMath.dylib /Users/rdeits/locomotion/drake-distro/build/install/lib/libBulletSoftBody.dylib /Users/rdeits/locomotion/drake-distro/build/install/lib/libfcl.0.6.0.dylib /Users/rdeits/locomotion/drake-distro/build/install/lib/libccd.2.0.dylib /usr/lib/libm.dylib /Users/rdeits/locomotion/drake-distro/build/install/lib/liboctomap.dylib /Users/rdeits/locomotion/drake-distro/build/install/lib/liboctomath.dylib /Users/rdeits/locomotion/drake-distro/build/install/lib/liboctomap.1.8.0.dylib /Users/rdeits/locomotion/drake-distro/build/install/lib/liboctomath.1.8.0.dylib multibody/shapes/libdrakeShapes.dylib multibody/joints/libdrakeJoints.dylib util/libdrakeGeometryUtil.dylib math/libdrakeMath.dylib common/libdrakeCommon.dylib /Users/rdeits/locomotion/drake-distro/build/install/lib/libgflags.2.2.0.dylib thirdParty/bsd/spruce/libspruce.dylib thirdParty/zlib/tinyxml2/libtinyxml2.dylib -Wl,-rpath,/Users/rdeits/locomotion/drake-distro/build/drake/multibody/parsers -Wl,-rpath,/Users/rdeits/locomotion/drake-distro/build/drake/multibody -Wl,-rpath,/Users/rdeits/locomotion/drake-distro/build/drake/multibody/collision -Wl,-rpath,/Users/rdeits/locomotion/drake-distro/build/drake/multibody/shapes -Wl,-rpath,/Users/rdeits/locomotion/drake-distro/build/drake/multibody/joints -Wl,-rpath,/Users/rdeits/locomotion/drake-distro/build/drake/util -Wl,-rpath,/Users/rdeits/locomotion/drake-distro/build/drake/math -Wl,-rpath,/Users/rdeits/locomotion/drake-distro/build/drake/common -Wl,-rpath,/Users/rdeits/locomotion/drake-distro/build/drake/thirdParty/bsd/spruce -Wl,-rpath,/Users/rdeits/locomotion/drake-distro/build/drake/thirdParty/zlib/tinyxml2 -Wl,-rpath,/Users/rdeits/locomotion/drake-distro/build/install/lib && :
Undefined symbols for architecture x86_64:
  "_PyBuffer_Release", referenced from:
      pybind11::buffer_info::~buffer_info() in rbtree.cc.o
  "_PyCFunction_NewEx", referenced from:
      pybind11::cpp_function::initialize_generic(pybind11::detail::function_record*, char const*, std::type_info const* const*, unsigned long) in rbtree.cc.o
  "_PyCFunction_Type", referenced from:
      pybind11::cpp_function::initialize_generic(pybind11::detail::function_record*, char const*, std::type_info const* const*, unsigned long) in rbtree.cc.o
```

(and so on). 

The issue appears to be that the library generated by pybind11 isn't linked against the Python library at all. Digging around inside the pybind11 code shows that that's exactly the case: https://github.com/pybind/pybind11/blob/7830e8509f2adc97ce9ee32bf99cd4b82089cc4c/tools/pybind11Tools.cmake#L104

That linking choice works fine for me when I start a bare CMake project, and it works fine inside the new version of IRIS that I just pushed, but something about the way Drake is built causes it to fail for us. Is Drake trying too hard to link that library right now instead of letting it resolve at import time? 

Is this something something rpath something?

I'd appreciate any input from people who actually understand how cmake works.

rdeits, 2017-01-27:
For reference, my code is here: https://github.com/rdeits/drake/tree/pybind-rbtree

rdeits, 2017-01-27:
It seems to be caused by this: https://github.com/RobotLocomotion/drake/blob/master/drake/CMakeLists.txt#L63

rdeits, 2017-01-27:
Another issue (I'll make a new super-issue with all of this later):

pybind11 ships its own `FindPythonNew.cmake`, which seems to work much better than the one shipped with cmake (for example, it actually works on OSX). Unfortunately, if python has already been found with a call to `find_package(Python)`, it looks like `FindPythonNew` isn't called (because cmake). This means that some variables that pybind11 relies on aren't set. The specific example I ran into is that `PYTHON_MODULE_EXTENSION` isn't set, so python modules don't have a `.so` suffix: https://github.com/pybind/pybind11/blob/7830e8509f2adc97ce9ee32bf99cd4b82089cc4c/tools/pybind11Tools.cmake#L89

david-german-tri, 2017-01-27:
> It seems to be caused by this: https://github.com/RobotLocomotion/drake/blob/master/drake/CMakeLists.txt#L63 which suggests that this is an intentional behavior. 

Definitely - no one wants to debug runtime link failures.

> Can we disable that flag just for the pybind11 bindings?

Seems reasonable to me.

mwoehlke-kitware, 2017-01-31:
Python bindings that use Python functions? Scandalous :wink:.

I remember noticing this playing with your "experiment" a wondering *why* pybind11's helper function doesn't just link to the Python library. Is there a problem just doing that?

> pybind11 ships its own `FindPythonNew.cmake`, which seems to work much better than the one shipped with cmake

Drake has its own `FindPython.cmake`, possibly we should use the pybind11 one and/or incorporate its improvements?

rdeits, 2017-01-31:
@mwoehlke-kitware I wondered that as well. The rationale is explained here: https://github.com/pybind/pybind11/blob/master/tools/pybind11Tools.cmake#L95

Basically, linking to a specific python library would make it impossible to load that library from a different python interpreter running the same python version.

rdeits, 2017-02-01:
@jwnimmer-tri 's suggestion of string substitution to apply the appropriate flags to the pybind11 targets seems to have worked, thanks!

jonathan-owens, 2017-02-16:
Looks like this is happening again: https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=394063

jamiesnape, 2017-02-16:
Looks like an IRIS issue. I will look into it. FYI @rdeits.

EricCousineau-TRI, 2017-08-02:
Should this be closed since IRIS is (unfortunately) no longer in Drake?

rdeits, 2017-08-03:
I...didn't know IRIS was gone. Ok, closing.

closed 2017-08-03 by rdeits

jwnimmer-tri, 2017-08-03:
FYI #6243 is the tracking issue to re-enable IRIS and use it.

As the CMake transition is wrapping up now, it might be worth adding IRIS to the (Bazel) build sooner rather than later, using some simple proof-of-life example in either C++ or Python.  That way we could fix up any build or link problems like the above at a suitable pace, versus perhaps having to rush it once #6243 work resumes.

/CC @stonier

rdeits, 2017-08-03:
Yeah, I knew the quadrotor code was gone, but didn't realize IRIS was entirely removed (since it was used by things other than quadrotor). Ok, thanks for the update.