RobotLocomotion/drake #4934: Monolane improvement

jwnimmer-tri, 2017-01-27:
Improve monolane to support segments with multiple lanes of constant lateral offset from a reference.  The only thing we care about is representing a highway on & off ramp single lane merging onto (off of) a trivial multi-lane highway (without any lane drops, etc.).

maddog-tri, 2017-01-30:
Here are my notes on what should be done to achieve this goal.
 * Implementation (`monolane::RoadGeometry` and its children):
   * The `Lane` base class and its descendents (`LineLane`, `ArcLane`) should be promoted to the `Segment` level (pretty much replacing the current mostly content-free `Segment` class).  The concept of "planar reference curve" will thus apply to the segment, and the segment will manage a surface with an (s,r,h) parameterization.
   * In addition to the reference curve, elevation polynomial, and superelevation polynomial, a `Segment` will have the parameters:
     * number of lanes
     * r-spacing between lane centers (which really should be a constant over whole RoadGeometry)
     * r-offset of Lane 0 (zero) from the segment's r=0 reference curve
     * (The gist of this is that the r=0 center of the lane with index j will be offset from the r=0 reference of a segment by `(j * spacing) + offset`.)
   * A `Lane` will now basically contain its index, and a lane's methods will call up to its segment with that index (which will be used to futz with the `r` parameters in the math).
     * Note well that the `length()` of each `Lane` belonging to a curved `Segment` will be different.  The `r` position parameters from one lane to another will be related by a constant offset no matter the shape, but the `s` parameters will be more exciting (e.g., for a circular arc reference curve, proportional to the difference in radius from one lane to next).
   * The nominal bounds (e.g., `lane_bounds()`) might as well just be the r-midpoint between two adjacent lanes.  For the min/max lane, however, there should be independent min-bound and max-bound.  In other words:  the "number of lanes is 1" case should reduce to the current `monolane` case.
   * Similarly, the driveable bounds (`driveable_bounds()`) should be a single min/max defined on a Segment (since the Segment represents the sheet of driveable asphalt), which is then translated appropriately into the r-frame of each Lane.
 * `Builder` and the YAML format:
   * No changes to the `Endpoint` class and corresponding `point:` language.
   * `Connection` has some new/modified semantics:
     * Needs a parameter specifying the number of lanes (e.g., could default to 1 in the YAML).
     * Instead of specifying "starts at the given `Endpoint`", needs to specify "Lane `j` starts at the given `Endpoint`"
     * (Displacement parameter is the same; implicitly, it will mean displacement to finish of Lane `j`.)
     * Needs a new parameter for the lateral offset of the 'segment reference curve' from Lane 0.
       * This is important even for straight/linear segments, because the segment's r=0 is the axis of rotation of the superelevation.
   * The `start()` and `end()` methods of `Connection` (and corresponding references in the YAML) will need `lane_index` parameter --- since the new meaning is "give me the start/end of lane `j` of the connection".
   * The `Builder` constructor needs a new lane-spacing parameter, and the semantics of `lane_bounds` and `driveable_bounds` will then be "bookend bounds" as discussed above.

That should do it.  I haven't looked at the actual math yet, but I think the `r`-translation from segment-frame to lane-frames should be pretty straightforward.  The true arc-length (max `s`) of a lane will be dicey, but a similar lame approximation as is already being performed will probably work just fine (for the time-being, as it does already, for the time-being).

jwnimmer-tri, 2017-03-02:
Upon further discussion, we may not need this right away.  Lowered priority.

jwnimmer-tri, 2017-05-17:
I'm not foreseeably going to work on this.  Kicking to @stonier for backlog.

maddog-tri, 2017-07-13:
"Is there any code previously written in an unmerged branch or should we start from scratch?" asked @basicNew.

I don't believe there is any existing work on this (besides the design work in this issue).  I'd suggest duplicating `maliput/monolane` into a `maliput/multilane` and starting there.  That way, this new `multilane` backend can be checked-in in phases without disturbing `monolane` users, and `monolane` can be subsequently deprecated and phased out.  (`dragway` might be phased out as well, since this should cover all of its functionality, too.)

maddog-tri, 2017-07-13:
Oh, and, if it were me, I would evolve the `monolane` code starting from the inside and working out.  In other words:  improve `Lane` and `Segment` first, and let the `Builder` just create single-lane instance at first.  Then improve `Builder` to have a multi-lane API... and then finally express the multi-lane features in the YAML at the very end.  This way, `multilane` can be PR'ed in those three stages, with all the existing unit-test in place at each stage (even if the later-stage tests only exercise single-lane functionality during earlier stages).

basicNew, 2017-07-19:
Completely agree with that and I think we may even need to split things into smaller chunks to avoid big PRs.

basicNew, 2017-07-20:
@maddog-tri this would be a proposed tasks decomposition, which we would map into separate issues, leaving this as a meta-issue to track the general progress. I don't think there is anything new here to what you already mentioned above, only a design tweak on the second option and the final `Builder` / YAML format approach. Please let me know what you think and if we should create these issues and start tackling them.

- [ ] Copy monolane directory into multilane, changing the required names (classes, variables, tests) from monolane to multilane. Verify of course that all tests pass (Ref: https://github.com/RobotLocomotion/drake/issues/4934#issuecomment-315194694).

- [ ] Merge the `Lane`/`LineLane`/`ArcLane` classes with the `Segment` class (which is almost empty), creating a new hierarchy (Ref: https://github.com/RobotLocomotion/drake/issues/4934#issuecomment-275997481). I wonder however if it wouldn't be better to have a concrete `Segment` class that has a `reference_curve` instance variable and then create a hierarchy of `SegmentGeometry`. This would allow us to work in smaller chunks (a first PR that adds the geometry hierarchy and a second one that performs the refactor), it would make it easier in the future to add new geometries while also being easier to test. Finally, we need to think about the asphalt bounds computation using a baseline curve + lane offsets, which I think should be promoted to the `Segment` class too.

- [ ] Properly implement number of lanes in a segment, removing the hardcoded value. Assume a fixed separation between them along the whole lane (Ref: https://github.com/RobotLocomotion/drake/issues/4934#issuecomment-275997481).

- [ ] Allow configurable r-spacing between lane centers (Ref: https://github.com/RobotLocomotion/drake/issues/4934#issuecomment-275997481). One thing we should define here is if this is going to be mandatory (i.e. it has to be specified when adding a lane) or optional (and if the latter, we should define a default value).

- [ ] Allow configurable r-offset of Lane 0 from the segment's r=0 reference curve (Ref: https://github.com/RobotLocomotion/drake/issues/4934#issuecomment-275997481). Same as above, we should defne here if this is going to be mandatory or optional (and if the latter, we should define a default value).

- About evolving the `Builder` and YAML format I would like to propose a different approach, splitting the changes into smaller chunks but doing them both for the `Builder` and the YAML format:
  - [ ] Specify the number of lanes in a connection.
  - [ ] Specify the r-spacing between lane centers (We may need some clarification here on how to specify this in the format - re: 'Instead of specifying "starts at the given Endpoint", needs to specify "Lane j starts at the given Endpoint"' - ).
  - [ ] Specify the r-offset of Lane 0.

maddog-tri, 2017-09-01:
@agalbachicar, here is feedback on this next step:

> Hi Matt. [Here](https://github.com/ekumenlabs/drake/tree/test/Multilane_RoadCurve_Lateral_Offset/drake/automotive/maliput/multilane) is the test branch with the next PR (I still need to rebase comments from previous PRs).
> 
> The main idea behind this PR is the following:
> 
> - A new interface is created: `C2ScalarFunction`. It provides the necessary interface to define what's necessary from the Lane's internal implementation of an elevation and superelevation function. Mainly the function evaluation and its derivatives.
> - `CubicPolynomial` inherits from `C2ScalarFunction` and changes its methods to be overriding the interface.
> - A new class is introduced, it is called `Elevation` (I was thinking of changing the name to `ComposedElevation` or similar). This function implements `C2ScalarFunction` and it is composed of both `CubicPolynomials` (for reference elevation and superelevation) as well as the offset distance `r` at which the `ComposedElevation` is evaluated. The idea behind this class is to keep the elevation composition transparent for `Lane` class as much as possible.
> - `RoadCurve` consumes a `ComposedElevation`/`Elevation` instead of a `CubicPolynomial`.
> - `RoadCurve` adds a new method, `Offset`. This method should provide the facilities to offset the reference curve (over z=0 plane) and to create an new `ComposedElevation` object based on the previous displacement.
> 
> So, all in all, the idea behind this is that a `Segment` handles a reference trajectory (`RoadCurve` implementation). Each new Lane will consume an offset of the reference trajectory at `r` distance what will ease the process of keeping several Lanes inside a Segment and manage the offset at the `Segment` level pushing that down to `Lane`'s implementation.
> 
> Up to now I have some tests for geometry and I was planning to make a sample implementation of the integration with the `Segment::NewLane()`. Something left out of the implementation is lane bounds and driveable bounds management. 

I don't think that the `C2ScalarFunction` notion will actually work, and the task at hand is actually pretty simple.

What should work:

  * (Don't add any kind of `Offset` to `RoadCurve`.)
  * Construct each `Lane` with an `r0` offset parameter.
  * In `Lane`, in the front-end methods which accept a `api::LanePosition` input parameter, offset the `r` component by `r0` --- e.g., `lane_pos.r()` becomes `lane_pos.r() + r0_`.
  * Likewise, subtract r0 when outputting a `LanePosition`.

I believe that all the math is linear in `r` in such a way that this works just fine (though I would want to triple-check this).

(The fundamental problem with the `C2ScalarFunction` and `ComposedElevation` approach is that one actually needs the individual pieces (planar reference curve, elevation function, superelevation function) in order to compute the API methods.  E.g., `ComposedElevation` cannot yield the complete `(s,r,h)` to `(x,y,z)` transformation.)

agalbachicar, 2017-09-04:
@maddog-tri there is only one thing I found not to be nicely coded. It is the `Lane::do_length()` method. It returns `road_curve_->trajectory_length()` but it will be referring to the `road_curve_` trajectory instead of the Lane's curve path integral. We can solve this situation by scaling the `RoadCurve::trajectory_length()` when `Lane::road_curve_` pointer  is of type `ArcRoadCurve` but it will imply using `dynamic_cast`.

Do you have any suggestions?

agalbachicar, 2017-09-04:
Related to the previous [comment](https://github.com/RobotLocomotion/drake/issues/4934#issuecomment-327008804), if the `Lane` is fed with an ArcRoadCurve , whenever calling `Lane::p_from_s()`, `s` should be scaled by the offset reference curve length instead of the base reference curve. Leaving that scale factor will return badly scaled `p` coordinates.

maddog-tri, 2017-09-05:
Hmm... good points.  All the non-linearity is hiding in the s<->p relationship!

Suggestions:
  * `RoadCurve::trajectory_length()` should become `trajectory_length(double r)`, taking an `r` parameter for a constant lateral offset.  `Lane::do_length()` thus calls it with the Lane's `r0_`.
  * Similarly, `Lane::p_from_s(double s)` needs to become `RoadCurve::p_from_s(double s, double r)`.

That's the easy part --- the tough part is implementing those functions.  An accurate answer is going to require numerical integration, but maybe there is some simple low-order analytical approximation that will be good enough for the short-term (e.g., like the existing `p_s()` and `s_p()` for `CubicPolynomial`).  I'll think about that more during daytime.

> whenever calling `Lane::p_from_s()` `s` should be scaled by the offset reference curve path length instead of the base reference curve path length

Note that this would only work for zero superelevation.  With non-zero superelevation, even LineRoadCurve gets *screwy* (ha-ha!).  Imagine a LineRoadCurve with constant zero elevation but linearly increasing superelevation:  at a non-zero offset `r0` from the center, the path is a corkscrew.

agalbachicar, 2017-09-05:
> Imagine a LineRoadCurve with constant zero elevation but linearly increasing superelevation: at a non-zero offset r0 from the center, the path is a corkscrew.

Nice example! There is no support for this at this moment. Am I right?

> That's the easy part --- the tough part is implementing those functions. An accurate answer is going to require numerical integration, but maybe there is some simple low-order analytical [...]

Maybe [this](https://bitbucket.org/ignitionrobotics/ign-math/src/a2c340e6d901ff065cfa828c0ebc395d712e9bc1/src/SplinePrivate.cc?at=default&fileviewer=file-view-default#SplinePrivate.cc-103:126) can help. It's part of the work @hidmic did to have better support on the Ignition' Splines for Maliput's RNDF implementation.

agalbachicar, 2017-09-13:
From PR #6985:

> (I anticipate NewLane() vanishing completely in a subsequent PR, replaced by additional parameters to Segment(...), once we actually handle multiple lanes. Segment will have to ensure that its lanes are indexed in the correct order of increasing r0.)

@maddog-tri  a couple of questions come to my mind:

- Should lanes be created inside `Segment`'s constructor?
- By doing so, how should `Lane`'s ID  be set?

maddog-tri, 2017-09-14:
Yeah, "vanishing completely" was probably a bit overzealous.

The free parameters for a Lane are `id`, `r0`, and `lane_bounds`.  (As discussed elsewhere, the `driveable_bounds` are derived from the `r0` and the Segment-wide bounds known to the Segment.) `id` can be any valid ID. `lane_bounds` can be anything bounded by the Lane's `driveable_bounds`.

`r0` is a little more complicated:  the Lanes in a Segment, as returned by `Segment::lane()` are required to be indexed in right-to-left order.  Each Lane also knows its index, and knows its left/right neighboring Lanes (as returned by `Lane::to_left()` and `Lane::to_right()`).

So, my original thought was to supply a list of `{id, r0, lane_bounds}` to the `Segment()` constructor, and have it figure out the correct ordering of the Lanes (I think just sorting on `r0` would do it?) and then construct const Lanes in the correct order with correct indices/etc from the very beginning.

Another option would be to insert/re-sort with each `NewLane()` call --- which would also entail potentially modifying the index values of Lanes created earlier. 

Another option would be to simply DRAKE_DEMAND that each call to `NewLane()` specifies a Lane to the left of the previous call.  That's not the most polite option, but definitely the most practical. I think that's the winner.

agalbachicar, 2017-10-06:
In PR #7171 `multilane::Builder` interface refactor became a new issue, #7194, which is being discussed. Consequently, the designed previously presented in this [comment](https://github.com/RobotLocomotion/drake/issues/4934#issuecomment-275997481) may be modified.

Just for the purpose to state current Multilane status, from PR #7127, the `RoadGeometry` is finished but it lacks of the following items:

- Support for elevation polynomials of higher order than linear.
- Support for superelevation when there is more than one lane.
- `Builder` interface.
- Loader utility methods and YAML format definition.
- Provide test coverage to `RoadCurve` methods that originally belonged to `Lane` class.

In order to finish Multilane, the following road map is proposed:

- Finish `Builder` design issue, #7194, and address the necessary changes.
- Adjust the YAML format and adjust loader utilities to the previous format and the new `Builder` interface.
- Remove constraints in elevation and superelevation polynomials by correctly (or as accurate as necessary) computing _s --> p_ and _p --> s_ relationships.
- Provide test coverage to `RoadCurve` methods.

agalbachicar, 2017-10-06:
@maddog-tri I would like to know your opinion on the proposed road map so as to be all aligned.

maddog-tri, 2017-10-10:
@agalbachicar Yes, I concur; that sounds like the right order in which to do things.

agalbachicar, 2018-02-08:
So, to give you all updates on the status of this issue, after merging PR #7626, a basic loader + builder is ready. They allow to construct `Segments` with `N` lanes starting from a specific `Endpoint` that belongs to `points` or use any of the `start` / `end` Endpoints of the other `Segment.Lanes`. This new `Segment` will end using `z_end` or `explicit_end` properties. [Here](https://drive.google.com/file/d/14D3eZmQtr8-BtWdF8rYVQcCAK9QKaZWt/view) you can find a draft of the yaml specification for `multilane`.

The following list describes the missing points of the implementation that needs to be tackled before having a complete `multilane`:

- [ ] `Loader` tests refactor to introduce GMock and evaluate calls to `Builder`.
- [ ] `Builder` support for `Connect` methods that allows to join lanes instead of references curves.
- [ ] `Loader` support for lane-to-lane methods provided by the `Builder`.
- [ ] Unblock constraints in `p <--> s` relation to properly integrate the arc-length of the lanes when superelevation is not constant and / or elevation polynomial degree is greater than linear (so far we support up to linear elevation and constant superelevation polynomials). This task can be solved by one of the followings:
  - [ ] Find an analytical solution.
  - [ ] Find a method that allows to numerically integrate (with a proper error constraint) `s=F(p, r)` and `p=G(s,r)` and is fast enough.
- [] Solve an issue in the elevation coordinate of `W(p, r, h)`.
- [] Create proper documentation of multilane's involved math.
- [] Create proper documentation of multilane's yaml specification.

liangfok, 2019-06-08:
@agalbachicar I'm assigning this to you since you're the author of the most recent comments listing things that need to be done for multilane to be complete. Please take a look at whether any of the listed tasks are still valid and, if so, migrate them as issues in Maliput's new repo. Once done, this ticket can be closed.

agalbachicar, 2019-06-10:
Closing this ticket as peding action items have already been migrated to downstream repository.

closed 2019-06-10 by agalbachicar