RobotLocomotion/drake #4940: Refactor AddLinearEqualityConstraint to use symbolic::Formula

soonho-tri, 2017-01-27:
We will have
```c++
Binding<LinearEqualityConstraint>
MathematicalProgram::AddLinearEqualityConstraint(const Formula& f)
```
and also the vector version.

So that we can write `AddLinearEqualityConstraint(x(0) + x(1) + x(2) == 3)` instead of `AddLinearEqualityConstraint(x(0) + x(1) + x(2), 3)`.

/cc @hongkai-dai

hongkai-dai, 2017-01-27:
I am thinking we can probably combine `AddLinearEqualityConstraint` and `AddLinearConstraint`. The difference between these two are the return type. Is it possible to find out whether the formula is an equality or inequality in the compile time? So that we can call
```C++
Binding<LinearEqualityConstraint>   binding1 = prog.AddLinearConstraint(2*x(0) + 3*x(1) = =2);
Binding<LinearInequalityConstraint> binding2 = prog.AddLinearConstraint(3*x(0) + 4*x(1) <= 3);
```

soonho-tri, 2017-01-27:
1. Isn't it the case that `AddLinearConstraint` takes both of lower-bound and upper-bound?
2. Do we have `LinearInequalityConstraint`?

hongkai-dai, 2017-01-27:
> Isn't it the case that AddLinearConstraint takes both of lower-bound and upper-bound?

Yes, I think it is easier for the user if he can write
```C++
AddLinearConstraint( 2 *x(0) + 3*x(1) <= 4);
```
rather than 
```C++
AddLinearConstraint( -std::numeric_limits<double>::infinity() <= 2 * x(0) + 3 * x(1) <= 4);
```
Namely if the linear inequality constraint is only enforce on the one side, then in the symbolic formula, we can ignore the other side.

Does `symbolic::Formula` allows us to write the linear inequality on both sides, like
```
1 <= 2 * x(1) + 3 * x(2) <= 4
```
?

> Do we have LinearInequalityConstraint?

No we do not have one, `LinearConstraint` is used for inequality constraint.

soonho-tri, 2017-01-27:
First of all, C++ doesn't support `1 <= 2 * x(1) + 3 * x(2) <= 4` syntax. We need to write:

```
(1 <= 2 * x(1) + 3 * x(2)) && (2 * x(1) + 3 * x(2) <= 4)
```

In this particular method, we can assume and make sure that the input formula is always in the form of `lb <= e && e <= ub` so that we don't need to run "decompose" method twice over `e`.

soonho-tri, 2017-01-27:
In general, I agree with you in that we should have a single method handling both of ineqs and eqs. I believe that this is the right direction which makes things more easier to use.

soonho-tri, 2017-01-27:
I think the question is "do we want both of the following or just choose one?"

1. `AddLinearConstraint(e, lb, ub)`
2. `AddLinearConstraint(lb <= e && e <= ub)`

hongkai-dai, 2017-01-27:
How about we supporting

1. `AddLinearConstraint(e, lb, ub)` for inequality constraint on both sides.
2. `AddLinearConstraint(e >= lb)` for greater than 
3. `AddLinearConstraint(e <= ub)` for less than.

soonho-tri, 2017-01-27:
BTW, @jwnimmer-tri just taught me http://en.cppreference.com/w/cpp/algorithm/clamp

soonho-tri, 2017-01-27:
It's C++17 feature but we can take this idea and I can implement
```
symbolic::Formula clamp(const symbolic::Expression& e, double lb, double ub);
```
and we can use it for `AddLinearConstraint(const symbolic::Formula)`.

hongkai-dai, 2017-01-27:
So when the user uses it, it will be
```
AddLinearConstraint(clamp(e, lb, ub));
```
? This looks more complicated to the user, than `AddLinearConstraint(e, lb, ub)`;

soonho-tri, 2017-01-27:
I was writing the same sentence...

soonho-tri, 2017-01-27:
@jwnimmer-tri , probably `clamp` doesn't save us much in this context?

jwnimmer-tri, 2017-01-27:
Up to you guys, you are the users.  I guess I was imagining that it was a way to reuse some existing base-case `AddConstraint` overload that take a bare Expression.

jwnimmer-tri, 2017-01-27:
But I see now you return this weird Binding stuff, so I guess nevermind.

soonho-tri, 2017-01-27:
OK.

@hongkai-dai , let's support the followings (basically what you said above + equality)

```
1. AddLinearConstraint(e, lb, ub) for inequality constraint on both sides.
2. AddLinearConstraint(e >= lb) for greater than
3. AddLinearConstraint(e <= ub) for less than.
4. AddLinearConstraint(e = c) for equality.
```

soonho-tri, 2017-01-27:
Wait.. but then what'd be the return type of `AddLinearConstraint`?

hongkai-dai, 2017-01-27:
Totally agree.

nitpick: the last one is `AddLinearConstraint(e == c)`, with two `=`?

hongkai-dai, 2017-01-27:
>Wait.. but then what'd be the return type of AddLinearConstraint?

Yes that is my question before, can we differentiate `e >= lb` and `e == c` in the compile time? Then we can  do SFINAE to return different types.

soonho-tri, 2017-01-27:
> can we differentiate e >= lb and e == c in the compile time?

No. Both of them return `Formula` type.

I think one solution is to have `LinearConstraintInequality` and `LinearConstraintEquality` as derived classes of `LinearConstraint`, and `AddLinearConstraint` returns `LinearConstraint`. But this is not so desirable solution since we need to check a return value at runtime and do a proper casting.

soonho-tri, 2017-01-27:
Or we merge `LinearConstraintInequality` and `LinearConstraintEquality` classes into a single class `LinearConstraint` but have a flag for its relational operator (`<=`, `==`) and use it in evaluation.

hongkai-dai, 2017-01-27:
OK, so how about we return `Binding<LinearConstraint>` for adding linear equality constraint? `LinearEqualityConstraint` is a derived class from `LinearConstraint`. So the API is
```C++
Binding<LinearConstraint> AddLinearConstraint(const Formula& f);
```
which supports adding all `<=`, `>=` and `==` linear constraint, but does not support the linear inequality constraint on both sides, and we still keep 
```C++
Binding<LinearConstraint> AddLinearConstraint(const Expression& e, double lb, double ub);
```

Inside `Binding<LinearConstraint> AddLinearConstraint(const Formula& f)`, it determines the type of the formula (either `>=`, `<=`, or `==`), and then call either `AddLinearConstraint(A, lb, ub, vars)` or `AddLinearEqualityConstraint(Aeq, beq, vars)` depending on the type of the formula.

soonho-tri, 2017-01-27:
Sounds like it's best we can do in the current design.

Here is a question. For now, any method taking `Binding<LinearConstraint>` has to check whether the argument is `Binding<LinearConstraint>` or `Binding<LinearEqualityConstraint>`. Is it correct?

soonho-tri, 2017-01-27:
or it doesn't matter since `LinearEqualityConstraint` sets both of `lb` and `ub` with the same value?

hongkai-dai, 2017-01-27:
I think inside the solver, the program needs to differentiate whether the linear constraint is inequality or not, since it can incur different type of optimization problems. For example, convex quadratic objective functions with linear equality constraints have an analytical solution, while there is no analytical solution for the same objective function with linear inequality constraint. 

But from the user side, linear inequality constraint and linear equality constraint is not so different, so it is OK to return a `LinearConstraint` object, although it is actually a linear equality constraint.

soonho-tri, 2017-01-27:
```
  template <typename DerivedA, typename DerivedB>
  LinearEqualityConstraint(const Eigen::MatrixBase<DerivedA>& Aeq,
                           const Eigen::MatrixBase<DerivedB>& beq)
      : LinearConstraint(Aeq, beq, beq) {}
```
Never mind. I just checked the source code.

soonho-tri, 2017-01-27:
I see. It's clear for me now.

soonho-tri, 2017-02-01:
https://github.com/RobotLocomotion/drake/pull/4972 closed this.

closed 2017-02-01 by soonho-tri