RobotLocomotion/drake #4942: quaternion.h quat2rotmat() normalizes the quaternion (when it shouldn't)

edrumwri, 2017-01-27:
quat2rotmat() normalizes quaternions on entry; the function specification indicates that the quaternion input should be normalized on entry, but then goes ahead and normalizes the quaternion anyway.

This subtle issue results in derivatives of functions that use quaternions to be computed incorrectly. For example, the time derivative of X*q is different than the time derivative of X*q/norm(q).

We expect this issue to be fixed automatically by the introduction of MultibodyTree.

amcastro-tri, 2017-01-27:
Thank you for noticing this issue @edrumwri. I also hope that when we start adding well documented implementations plus unit tests we will get rid of this kind of problems. 
I am not working on the new quaternion implementation yet but I will for sure ping you on this when I start that work.

edrumwri, 2017-01-27:
Noticing = several hours of work to find out why an existing test wasn't passing with new PR!

amcastro-tri, 2017-02-01:
We have Eigen's `Quaternion<T>::toRotationMatrix()`. I wonder if it has the same issue.

sherm1, 2017-02-01:
No, Eigen's method doesn't normalize so it is OK.

jwnimmer-tri, 2018-12-21:
@mitiguy Has this issue been fixed already?

mitiguy, 2018-12-22:
quat2rotmat() has been deprecated.
This issue can be closed.

DRAKE_DEPRECATED("Use  math::RotationMatrix(Eigen::Quaternion). "
                 "Code will be deleted after February 5, 2019.")

closed 2018-12-22 by mitiguy