RobotLocomotion/drake #4946: Bitten by git submodules again

edrumwri, 2017-01-27:
I've been bitten yet again by git submodules for external dependencies. I'd already adjusted my workflow to carefully step around this issue, but it seems that merging the master branch into my PRs has the nasty tendency to activate this problem. For those unaware of the problem, please take a look at the latest status of my branch #4604 or consider a time where you may have accidentally committed SHA changes to an external submodule.

Yes, I could and can modify my workflow yet again to accommodate this, or I could up my level of gitfu, but really, why is this landmine here and why is it possible for this problem to occur? Is it even common behavior to modify the external SHAs?

jwnimmer-tri, 2017-01-27:
Yes, the sha's change not-infrequently, though in particular this week they have been changing more than usual.

Once we are all-Bazel and CMake-only-as-a-wrapper we are not going to use git submodules anymore.  So I guess for now I would advocate better-documenting the fixups and workarounds, rather than improving the core technology related to submodules.  (But of course if there are easy non-doc fixes to submodules that would be great -- I just don't know of any.)

sherm1, 2017-01-27:
So Bazel will fix our submodule problems! Ummm ... any chance it can grow hair on my head too?

edrumwri, 2017-01-27:
Halleleujah!

david-german-tri, 2017-01-30:
Submodules were added in #2487, replacing a glob of CMake code that managed the externals.  That code was an even bigger landmine.

Given that a CMake core developer recommended that we use submodules instead of CMake, I agree with @jwnimmer-tri that there's probably no low-hanging fruit here, and we should focus on the switch to Bazel.

jwnimmer-tri, 2017-01-30:
So the recovery procedure is something like https://github.com/RobotLocomotion/drake/pull/4805#issuecomment-274662451:
 1. Level 3 nuke from #4551.
 1.    Re-run cmake.
 1.    Re-run download-all.
 1.    git checkout the sha we want (as found on https://github.com/RobotLocomotion/drake)
 1.    git commit

Closing this as a duplicate of #4551.

Ask on Slack if you have more problems, and David or I will help with specific problems.

closed 2017-01-30 by jwnimmer-tri