RobotLocomotion/drake #4947: Advising users to install libgl1-mesa-dri is a bad idea

jwnimmer-tri, 2017-01-27:
Unconfirmed (but trustworthy reports) state that:
> The director-recommended package on the http://drake.mit.edu/ubuntu_trusty.html page "libgl1-mesa-dri" is tied to the trusty-original HWE stack, so unless you installed from original 14.04 media (and didn't upgrade HWE) it's going to break your install, where "break" = "uninstall xorg without replacing it".

liangfok, 2017-01-28:
I just tried to install `libgl1-mesa-dri` on a "dirty" Ubuntu 14.04.5 machine ran into the following error:

```
$ sudo apt-get install libgl1-mesa-dri
[sudo] password for liang: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 unity-control-center : Depends: libcheese-gtk23 (>= 3.4.0) but it is not going to be installed
                        Depends: libcheese7 (>= 3.0.1) but it is not going to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```
It's a relief the installation attempt did not corrupt my display server. I'll have to test again using a clean  Ubuntu Trusty before resolving this issue.

jamiesnape, 2017-01-31:
Also, the average user does not need it anyway, since it is only installed for the benefit of xvfb.

jamiesnape, 2017-02-17:
@BetsyMcPhail Can you pre-install `libgl1-mesa-dri` and `xvfb` on the unprovisioned Xenial images and then remove the two packages from `install_prereqs.sh` and the Trusty install instructions?

That would fix this issue and #5187.

mwoehlke-kitware, 2017-02-17:
> Also, the average user does not need it anyway, since it is only installed for the benefit of xvfb.

What provides GL on newer trusty / xenial?

BetsyMcPhail, 2017-02-17:
Unprovisioned image has been updated and the packages have been removed from the Xenial script and Trusty instructions.

closed 2017-02-17 by BetsyMcPhail