RobotLocomotion/drake #4950: Python API cleanup

rdeits, 2017-01-28:
In porting the python bindings to pybind11, it became pretty clear that the API could use some cleanup. A few issues that I'd like to address:

- [ ] remplace "inverseKinSimple" and the like with lambdas inside the pybind files
  * these functions exist only to make SWIG happy. We don't need them cluttering up the source files
- [ ] move kRollPitchYaw etc. inside their enum class instead of being at the top level
  * just a minor nitpick, but there's no reason for them to be module-level values. They should be scoped inside their enum class just like they are in c++
- [ ] autodiff methods should accept and return numpy array of autodiffs
  * specifically, this means:
    * deleting the `toAutoDiff` function and the `values()` and `derivatives()` methods of the opaque `VectorXAutoDiffX` type. Instead, users can just use a numpy array of autodiffscalars just like they use an Eigen matrix of autodiffscalars.
- [ ] be consistent about which methods accept "points" and require that they be passed as 3xN numpy arrays (not 1D vectors)
  * some of the IK constraints take in points (such as points in a particular frame that will be constrained), as does RigidBodyTree::transformPoints. We should be consistent about whether those methods accept a [3 x N] matrix or whether a 1D vector of length 3 is also an acceptable input. 
	- [ ] deprecate the custom constructors that take 1D vectors in ik.py

Does this all seem reasonable?

patmarion, 2017-01-30:
Note: Director uses this API via director.pydrakeik, and it works for both drake/master and openhumanoids's aging drake fork.  There is a compatibility layer to fix api differences between those two versions of drake.  I think the compatibility layer was recently broken for the oh fork of drake, and I've yet to update it.  If the api continues to change further, I may just drop support for pydrakeik on oh's drake fork.

rdeits, 2017-01-30:
Of these, I see the most pressing change being the one about autodiff methods accepting and returning a numpy array of autodiff instead of a special wrapper type. That change will also give us consistency in the way python passes and receives `drake::Variable`s, `double`s, and `AutoDiffScalars`. 

Fortunately, I think that's also the least disruptive change. As far as I know, no one is using pydrake with autodiff yet.

RussTedrake, 2017-01-31:
reminder that we have one more anticipated change in the autodiff types coming (soonish):  doing the fixed-size chunking instead of variable size.  not sure if that impacts your decisions.

rdeits, 2017-01-31:
Good point, and I'm not sure exactly how that will affect things. Do we have a plan for how to provide an easy way to use the chunked AD types? 

For example, right now, if I want the jacobian of a point w.r.t. q, I can call `doKinematics` with q as a vector of AutoDiffMax73 and then pull out the derivatives. With the fixed-size chunking, I have to call `doKinematics` N times with chunks of size M and then assemble the entire gradient. Is there a plan to wrap a higher-level API around this?

RussTedrake, 2017-01-31:
Yes -- plan is to have helper functions which setup and pull out the chunked gradients.

rdeits, 2017-01-31:
Ok. I *think* that shouldn't affect the python bindings. There are two ways python can deal with AD:

1. Have the python API actually pass wrapped AD arrays down to Drake and back. Write a `jacobian()` function in python that does the chunking and extracts the result
2. Have python call a Drake `GetJacobian()` method of some kind

I'd actually prefer option (1), even though it means more python code. The reason is that, if implemented well, Drake's Eigen AD could also be a nice way to differentiate pure-python code that doesn't use any of Drake's kinematics or dynamics methods, or code that does some work in python and some work in Drake. Option 1 makes that easy, and option 2 doesn't. This is independent of whatever higher-level API drake uses in C++. 

I'm open to discussion about that if other people disagree. 

If we go with option 1, then I can write a simple python layer that uses Drake AD under the hood but doesn't expose it to the user. That way we can go forward with the changes I'm proposing, and later we can switch out to chunk-mode later without any API changes.

rdeits, 2017-01-31:
I've implemented option 1 as a demo on the `pybind-rbtree` branch as a demo. Here's an example of usage: https://github.com/RobotLocomotion/drake/pull/4949/files#diff-fb5025ee83d9b7dd582bd89aa251bd2fR35

With this interface, it's possible to move to chunked-mode AD in the background without any user-facing changes.

rdeits, 2017-02-13:
Autodiff cleanup is done. I've also created the `pydrake.forwarddiff` package which uses Eigen Autodiff and provides high-level wrappers to take derivatives, gradients, and jacobians of drake functions and basic python functions.

jwnimmer-tri, 2017-03-11:
In order to assign every issue and owner, I'm putting @rdeits on this one, but of course others are still invited to contribute.

EricCousineau-TRI, 2018-02-02:
Additional items:

- [ ] Move `rbtree_py` and `parsers_py` into `pydrake.multibody` (#7900)
- [ ] Move `ik` into `pydrake.multibody`

EricCousineau-TRI, 2018-02-02:
@rdeits Would it be OK if I roll `forwarddiff` into `autodiffutils`, and rename that lib to `autodiff`, and move that into `pydrake.common`? (I'd follow a deprecation process, so it wouldn't break anything for 2-6 months).

rdeits, 2018-02-02:
Yes, that seems perfectly reasonable.

EricCousineau-TRI, 2018-12-08:
Some items handled a while back (binding of `Eigen::Isometry3`, some moves), and future code seems to have addressed these (e.g. `pydrake.math.RigidTransform`); since other items are relevant to `attic/` items, will close this for now.

closed 2018-12-08 by EricCousineau-TRI