RobotLocomotion/drake #4955: Compiler Warning: "using integer absolute value function 'abs' when argument is of floating point type"

liangfok, 2017-01-29:
# The Warning

On OSX 10.11.6 using clang Apple LLVM version 7.3.0, the following warning occurred while building Drake: 

```
[101/253] Building CXX object multibody/benchmarks/cylinder_torque_free_analytical_solution/te...les/cylinder_torque_free_analytical_solution.dir/cylinder_torque_free_analytical_solution.cc.o
/Users/liang/dev/drake-distro-4/drake/multibody/benchmarks/cylinder_torque_free_analytical_solution/test/cylinder_torque_free_analytical_solution.cc:119:31: warning: using integer absolute value function 'abs' when argument is of floating point type [-Wabsolute-value]
  const double quat_epsilon = abs(1.0 - quat.norm());
                              ^
/Users/liang/dev/drake-distro-4/drake/multibody/benchmarks/cylinder_torque_free_analytical_solution/test/cylinder_torque_free_analytical_solution.cc:158:7: note: in instantiation of function template specialization 'drake::benchmarks::cylinder_torque_free_analytical_solution::TestQuaternionDtConstraintFromQuaternionDt<double>' requested here
      TestQuaternionDtConstraintFromQuaternionDt(quat_AB, quatDt);
      ^
/Users/liang/dev/drake-distro-4/drake/multibody/benchmarks/cylinder_torque_free_analytical_solution/test/cylinder_torque_free_analytical_solution.cc:613:7: note: in instantiation of function template specialization 'drake::benchmarks::cylinder_torque_free_analytical_solution::CalculateAngularVelocityExpressedInBFromQuaternionDt<double>' requested here
      CalculateAngularVelocityExpressedInBFromQuaternionDt(
      ^
```

sherm1, 2017-01-29:
Liang, see PR #4956 -- I don't know how to verify that it is fixed on the platform you described above. That doesn't seem to be run during CI.

liangfok, 2017-01-29:
I just noticed that this warning shows up in CI here:

https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-clang-ninja-continuous-debug/1303/consoleFull

PR #4956 (2017-01-30): Add a missing `using` to get the floating abs on OSX
A missing `using std::abs` caused OSX to choose the integer absolute value rather than floating.

Fixes #4955.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4956)
commit 0365c619 (2017-01-30): Add a missing "using" to get the floating abs on OSX (fixes #4955).
closed 2017-01-30 by sherm1