RobotLocomotion/drake #4971: We need a Drake document describing what AutoDiffScalar does & how to use it properly

sherm1, 2017-01-31:
Eigen's AutoDiffScalar has such an opaque interface that even those of us who have used it successfully can't remember how it works the next time we go to do something with it. It would be very helpful to have a Drake tutorial somewhere (Doxygen?) on this topic, perhaps illustrated with some unit tests.

From discussion in #4959 [here](https://reviewable.io/reviews/robotlocomotion/drake/4959#-KblFKj4CxYVPOJfoOjH).

@edrumwri threatened to write this in a moment of AutoDiffScalar-induced frustration, so I'm assigning this to you for now Evan!

sherm1, 2017-03-11:
@edrumwri, I'm reassigning this to you in case you are back in the mood. If not, I think you might as well close this issue.

edrumwri, 2017-03-11:
I'm no longer aggravated by this. I still think the issue is important, but
since the expected use of AutoDiff in drake does not seem to be set in
stone, I don't think it can be addressed right now.

--------------------------------
Evan Drumwright
Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

closed 2017-03-11 by sherm1