RobotLocomotion/drake #4990: Implement a Lane Change Demo Using Hard-Coded Drag Way

liangfok, 2017-01-31:
This will build off of #4989 .

jwnimmer-tri, 2017-01-31:
Possibly the MOBIL lane-change model is a good candidate, to complement the IDM car model.

jwnimmer-tri, 2017-02-01:
... and for an earlier spiral of this, even multi-lane bonehead cars + SimpleCar ego car under key/joy control driving on the #4989 would be a big help.  (In other words, we don't need agent planners in the first spiral.)

jwnimmer-tri, 2017-02-09:
Let's keep this issue about just the first spiral -- having a demo up and running, with either bonehead or IDM cars.  The SimpleCar can drive around and lane-change, but the agents don't have to.

I've filed #5113 about MOBIL as a possible second step.

liangfok, 2017-02-13:
Now that #5038 is merged, I plan on starting to work on this. My plan is to either use or mimic an [EndlessRoadCar](https://github.com/maddog-tri/drake/blob/c8679ca8ac2b60cc25b779c3b48eb0eda1ef6add/drake/automotive/endless_road_car.h).

liangfok, 2017-02-13:
Since #5032, which contains [EndlessRoadCar](https://github.com/maddog-tri/drake/blob/c8679ca8ac2b60cc25b779c3b48eb0eda1ef6add/drake/automotive/endless_road_car.h), has not been merged and I'm unable to compile the branch, I'm now looking to see whether I can simply plop a `SimpleCar` on top of a Dragway.

Here is the compile-time error I'm seeing when comiling @maddog-tri's "maliput-demo-carbits" branch:

```
[289/652] Building CXX object automotive/CMakeFiles/drakeAutomotive.dir/maliput/monolane/lane.cc.o
FAILED: /usr/bin/g++-4.9   -DEIGEN_MPL2_ONLY -DHAVE_SPDLOG -DdrakeAutomotive_EXPORTS -I/home/liang/dev/drake-distro-2/drake/.. -I/home/liang/dev/drake-distro-2/drake/pod-build/lcmgen -isystem /home/liang/dev/drake-distro-2/build/install/include/eigen3 -isystem /home/liang/dev/drake-distro-2/build/install/include -Werror=all -Werror=ignored-qualifiers -Werror=overloaded-virtual -Werror=extra -Werror=return-local-addr -Wno-unused-parameter -Wno-missing-field-initializers -g -fPIC   -std=gnu++14 -MMD -MT automotive/CMakeFiles/drakeAutomotive.dir/dev/endless_road_oracle.cc.o -MF automotive/CMakeFiles/drakeAutomotive.dir/dev/endless_road_oracle.cc.o.d -o automotive/CMakeFiles/drakeAutomotive.dir/dev/endless_road_oracle.cc.o -c /home/liang/dev/drake-distro-2/drake/automotive/dev/endless_road_oracle.cc
/home/liang/dev/drake-distro-2/drake/automotive/dev/endless_road_oracle.cc: In function ‘void drake::automotive::internal::IndexJunctions(const std::vector<drake::automotive::internal::SourceState>&, const std::vector<std::vector<drake::automotive::internal::PathRecord> >&, std::map<const drake::maliput::api::Junction*, std::vector<drake::automotive::internal::TimeBox> >*, std::map<int, std::vector<const drake::maliput::api::Junction*> >*)’:
/home/liang/dev/drake-distro-2/drake/automotive/dev/endless_road_oracle.cc:474:51: error: no matching function for call to ‘drake::automotive::internal::TimeBox::TimeBox(<brace-enclosed initializer list>)’
                                        s_in, s_out};
                                                   ^
/home/liang/dev/drake-distro-2/drake/automotive/dev/endless_road_oracle.cc:474:51: note: candidates are:
In file included from /home/liang/dev/drake-distro-2/drake/automotive/dev/endless_road_oracle.cc:2:0:
/home/liang/dev/drake-distro-2/drake/../drake/automotive/dev/endless_road_oracle-internal.h:62:8: note: drake::automotive::internal::TimeBox::TimeBox()
 struct TimeBox {
        ^
/home/liang/dev/drake-distro-2/drake/../drake/automotive/dev/endless_road_oracle-internal.h:62:8: note:   candidate expects 0 arguments, 6 provided
/home/liang/dev/drake-distro-2/drake/../drake/automotive/dev/endless_road_oracle-internal.h:62:8: note: constexpr drake::automotive::internal::TimeBox::TimeBox(const drake::automotive::internal::TimeBox&)
/home/liang/dev/drake-distro-2/drake/../drake/automotive/dev/endless_road_oracle-internal.h:62:8: note:   candidate expects 1 argument, 6 provided
/home/liang/dev/drake-distro-2/drake/../drake/automotive/dev/endless_road_oracle-internal.h:62:8: note: constexpr drake::automotive::internal::TimeBox::TimeBox(drake::automotive::internal::TimeBox&&)
/home/liang/dev/drake-distro-2/drake/../drake/automotive/dev/endless_road_oracle-internal.h:62:8: note:   candidate expects 1 argument, 6 provided
```

jwnimmer-tri, 2017-02-13:
@liangfok Are you able to make progress?

I think EndlessRoadCoar gloms together too many concerns at once.  I bet you could steal just the bonehead variant of it and get really far (+ SimpleCar) on a dragway.  Or also the TrajectoryCar should be able to drive the dragway okay also (just create a single-segment path to feed it, that is manually lined up with the dragway lanes).

liangfok, 2017-02-13:
> Are you able to make progress?

I'm still studying the code; not sure yet whether I'm blocked. One concern I have is there's nothing in `master` that demonstrates how a `maliput::api::RoadGeometry` is transformed into a "world" in which any vehicle model runs. Maybe I just missed it?

> I think EndlessRoadCar gloms together too many concerns at once. I bet you could steal just the bonehead variant of it and get really far (+ SimpleCar) on a dragway.

I looked through #5032 and wasn't able to identify where the bonehead variant of it is defined. Can you provide a pointer?

> Or also the TrajectoryCar should be able to drive the dragway okay also (just create a single-segment path to feed it, that is manually lined up with the dragway lanes).

Do you mean take `maliput::api::RoadGeometry` and convert it into a `automotive::Curve2`, which is needed by [TrajectoryCar's constructor](https://github.com/RobotLocomotion/drake/blob/7854cad01aaa718ec8abfec36fde90ac2ee1a2d4/drake/automotive/trajectory_car.h#L38)?

Thanks for any help!

jwnimmer-tri, 2017-02-13:
> One concern I have is there's nothing in master that demonstrates how a maliput::api::RoadGeometry is transformed into a "world" in which any vehicle model runs.

For viz you can add a flat plane manually if you want.  But otherwise, none of the Systems we're talking about needs RBT or a world at all.

> I looked through #5032 and wasn't able to identify where the bonehead variant of it is defined. Can you provide a pointer?

The kNone control type.

> Do you mean take maliput::api::RoadGeometry and convert it into a automotive::Curve2, which is needed by TrajectoryCar's constructor?

Nope.  It's a dragway.  The bonehead cars can drive straight.  It should be very easy to make a Curve2 with a single segment that aligns with the lane center (presumably we can just ask Maliput for the lane start and end in Geo (cartesian) space and then feed those into the Curve2 constructor.

liangfok, 2017-02-13:
Alright, my plan is to create an executable that:

1. Instantiates a `drake::maliput::dragway::RoadGeometry`.
2. Automatically converts the `RoadGeometry` into a `Curve2` that spans the entire driveable area of the dragway.
3. Instantiates a `TrajectoryCar` using `Curve2` for each lane that simply drives down the length of the dragway.
4. Loads the `TrajectoryCar` instances into an `AutomotiveSimulator`.
5. Runs the simulator.

Does the above sound right?

jwnimmer-tri, 2017-02-13:
That's okay to start.  We'll also want a SimpleCar (keyboard control) in there as well at some point.

Can it just be another mode of `automotive_demo`, instead of a separate program?

liangfok, 2017-02-13:
> Can it just be another mode of automotive_demo, instead of a separate program?

Sure. I will try to make it so.

jwnimmer-tri, 2017-02-14:
>  I'm unable to compile the branch (see below) ...

(Jumping back a bit.)  For the record, CI agrees with you that the PR doesn't compile.

maddog-tri, 2017-02-14:
W.r.t. the PR not compiling, meh, works fine on xenial.  (And, it is fixed for you and CI now, too.  FWIW, it *did* compile in CI until, in response to review comments, I added `{}` default initializers to the fields of `TimeBox` which apparently caused it to become non-aggregate in the eyes of the older gcc, etc, etc, and I did not notice the subsequent CI failure, because why would I... I'm busy looking at reviewable.io, not github.)

Also, I have a `maliput-demo-autobits` branch which bears the subsequent changes to `automotive-demo` and friends to run the demo.  (Does it work in CI?  Beats me.)

liangfok, 2017-02-14:
@maddog-tri, thanks for fixing the bug. I will look into `maliput-demo-autobits`. Regarding the CI failure, are you saying it didn't show up in Reviewable.io's donut-shaped icon at the top (see screenshot below)?

[image]

PR #5185 (2017-02-21): Adds a 10-Lane Dragway Demo With 10 TrajectoryCar Vehicles and One SimpleCar Vehicle.
Resolves #4990.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5185)
closed 2017-02-21 by liangfok