RobotLocomotion/drake #5012: Deprecation timeline for Ubuntu Trusty 14.04

jwnimmer-tri, 2017-02-01:
We should consider and publish Drake's deprecation and removal timeline for supporting Ubuntu Trusty 14.04.

We shouldn't (and don't want to) drop support immediately, but we should come to consensus on a reasonable estimated timeline for doing so, to help inform our engineering planning.

jwnimmer-tri, 2017-02-01:
My starting proposal would be something like this:
- Ubuntu 14.04 non-ROS builds are supported through June 2017, and thereafter unsupported.
- Ubuntu 14.04 ROS builds (using Catkin & GCC 4.9) are supported through the end of 2017, and thereafter unsupported.

jwnimmer-tri, 2017-02-21:
Actually, the GCC PPA that we already use on Trusty appears to offer even newer versions of GCC.  Possibly it would be enabling to use GCC 5.4 for the July..December runway of ROS-only support, if it was practical to switch the Catkin builds to using it.  I don't know off the top of my head if other Trusty-based ROS or HSR C++ libraries have any C++ABI incompatibilities with the newer `libstdc++` or GCC versions.

rdeits, 2017-03-03:
One additional thought: basically everything I do uses Travis for CI, so when I create an external project which uses Drake, I will also run that project on Travis. Since Travis still doesn't support 16.04 yet, using Drake post-Trusty will mean not using Travis, which probably means not having CI at all. That would be unfortunate.

david-german-tri, 2017-03-07:
from f2f: @patmarion says he's had success using 16.04 on Travis with Docker, using Docker files that @rdeits wrote.

rdeits, 2017-03-07:
Yup, also an option for sure. Setting up docker for Director was moderately annoying (at least a day of fiddling around with build systems), but it's probably better than not having CI.

stonier, 2017-04-03:
@jwnimmer-tri who were the groups that were still using Trusty? I'll connect with them and start investigating the status/roadmap.

jwnimmer-tri, 2017-04-03:
Great! I will respond offline.

stonier, 2017-07-23:
Relevant parties (location-robot-contact people) and whether they are ok with freezing now (20170723) and working off a branch with limited merging between master and the frozen branch:

* [x] CAM-HSR-@kunimatsu-tri : HSR will move to Xenial in 201710, fine with freezing now.
* [ ] HQ-Fetch-@naveenoid : 
* [ ] MIT-???-@RussTedrake
* [ ] Spartan-Various-@sammy-tri :

@sammy-tri / @patmarion are you able to vouch for spartan users?

@RussTedrake who would be a good spokesperson for developers at your lab?

@jwnimmer-tri other groups we should consider?

RussTedrake, 2017-07-23:
@peteflorence -- could you poll folks in lab and see if anyone is using / has a good reason for using Trusty?

jwnimmer-tri, 2017-07-24:
> @jwnimmer-tri Other groups we should consider?

Nope, I don't have anything else specific.

stonier, 2017-07-24:
> Could you poll folks in lab and see if anyone is using / has a good reason for using Trusty?

Note : would be perfectly fine to keep using Trusty at the point it is frozen. If at some time you'd like to take advantage of the new and shiny coming into master, then you'd have to jump.

rdeits, 2017-07-24:
The systems that will be the hardest to update are the ones on the robots. Atlas's onboard machines are all running Trusty, and I think Val's are too. I'm slightly nervous about upgrading the OS on those computers.

stonier, 2017-07-24:
> The systems that will be the hardest to update are the ones on the robots. Atlas's onboard machines are all running Trusty, and I think Val's are too. I'm slightly nervous about upgrading the OS on those computers.

As I mentioned, the trusty release will still be there, but not advancing (except perhaps the odd bugfix). Do you need the evolving tip of Drake on those systems?

RussTedrake, 2017-07-24:
good call @rdeits .  this is a tough one.  we are ramping back up on our experiments on those systems now.  yes, definitely want them tracking master; in fact, they are likely to be pushing master.

stonier, 2017-07-25:
Relevant parties (location-robot-contact people) and when they are amenable to freezing a Trusty branch with the expectation that their would be limited merging between master and Trusty branches and either none/limited CI support.

* [x] CAM-HSR-@kunimatsu-tri : HSR will move to Xenial in 201710, fine with freezing now.
* [x] TRI-CAR-@stonier : trusty support now required until post SFD, i.e. ~20171001
* [ ] HQ-Fetch-@naveenoid : Fetch still using Trusty, uncertain yet whether they need Drake on Fetch
* [ ] HQ- Remy-@naveenoid : 
* [x] MIT-Atlas-@RussTedrake/@rdeits : Atlas still using Trusty, they are ramping up experiments again
* [ ] Spartan-Various-@sammy-tri :

@sammy-tri / @patmarion are you able to vouch for spartan users?

patmarion, 2017-07-25:
I'm fine with making Spartan Xenial only but we should check with other users.

sammy-tri, 2017-07-25:
I'm also fine with making Spartan Xenial only, but I can only speak for users inside of TRI.

stonier, 2017-07-30:
Relevant parties (location-robot-contact people) and when they are amenable to freezing a Trusty branch with the expectation that their would be limited merging between master and Trusty branches and either none/limited CI support.

* [x] CAM-HSR-@kunimatsu-tri : HSR will move to Xenial in 201710, fine with freezing now.
* [x] TRI-Driving-@stonier : Trusty support required until post SFD, i.e. ~20171001
* [ ] HQ-Fetch-@naveenoid : Fetch still using Trusty, uncertain yet whether they need Drake on Fetch
* [ ] HQ- Remy-@naveenoid : 
* [x] MIT-Atlas-@RussTedrake/@rdeits : Atlas still using Trusty, they are ramping up experiments again
* [x] Spartan-Various-@sammy-tri / @patmarion : ok tentatively, pending input from other spartan user groups

stonier, 2017-07-30:
At this point, there looks to be provisional support to drop (fully or partially) Trusty support in October 2017. Actions between now and then:

1. Confirm the decision
2. @naveenoid follow up with the HQ robotics groups, but at this point I'd say they are not locked in yet and can plan to be forward facing, i.e. Drake use cases for Xenial only. Actually, they almost have to as complete ROS compatible support for Drake on Trusty is beyond the resources we have right now.
3. @RussTedrake, @deits, ... will explore options. If you can move to Xenial, that will be a win for everybody (lab included I suspect). Let's review early September.

RussTedrake, 2017-07-30:
Just to be sure i was clear -- some of the robot apis are on trusty and are black boxes.  I don't anticipate any troubles, and promised that we'd ask and/or try rolling forward all of the bots, but i don't feel like we can lock in a timeline until we learn a little more.

stonier, 2017-07-31:
I should probably have put 'Confirm the decision' at 3 to avoid the confusion. Holding off for now until some time in September - is it likely the lab will have time to explore the problem before then? Am hoping we can review the problem and put in place a plan for going forward by October.

rdeits, 2017-08-03:
Update from the Val slack: the current blocker appears to be that upgrading Val to 16.04 implies upgrading to ROS Kinetic, and the `orcos_toolchain` package is not currently built for Kinetic.

stonier, 2017-08-06:
I suspect you mean the `orocos_toolchain`?

rdeits, 2017-08-06:
Good catch, thanks. I've edited the comment.

stonier, 2017-08-07:
Ach, sort of awkward with it's custom set of instructions, but it looks like it should be reasonably easy to upgrade and some have had success with it, e.g. [here](http://rtt-lwr.readthedocs.io/en/latest/install/install-16.04-kinetic.html#).

This might be something I can quickly help with if you need it. Just reach out if so.

rdeits, 2017-08-07:
Ok, thanks!

stonier, 2017-09-25:
Proposing to freeze Drake Trusty support on October 6th with the following process:

* [ ] Oct 6: release branch for Trusty that will assist migration or for continued use in legacy builds
* [ ] Oct 6: updated documentation
* [ ] Oct 6: winding down binary releases and shambhala instructions for Trusty
* [ ] Oct 6 - Nov 1: providing support only for critical bugfixes on the Trusty release branch
* [ ] Nov 1: winding down CI jobs for Drake on Trusty and dropping support for the release branch

The updated documentation and shambhala support may take a few extra days to realise.

Continued support and bugfixing for Trusty will cease from November onwards due to the lack of CI coverage and available human resources available.

Feedback welcome.

jamiesnape, 2017-10-03:
https://github.com/RobotLocomotion/drake/tree/last_sha_with_trusty_support

jamiesnape, 2017-10-03:
Trusty support has been removed from `master`. Bug fixes should be merged to the maintenance branch [`last_sha_with_trusty_support`](https://github.com/RobotLocomotion/drake/tree/last_sha_with_trusty_support).

closed 2018-04-03 by jwnimmer-tri