RobotLocomotion/drake #5013: MultibodyTree features requests.

amcastro-tri, 2017-02-01:
This issue is opened to track a number of issues with RBT that need to be taken into consideration in the design of `MultibodyTree`. This issue will therefore help close those old `RBT` issues without keeping track of them.

- [ ] Compute Rigid Body Inertias with respect to `Bo` rather than `Wo`. #4621.

kunimatsu-tri, 2017-03-17:
I have a request for MultibodyTree, which is related to #3088.

I'm working on "label image" output feature of RGBD camera simulator.
The "label image" contains an unique ids associated to an object in each pixel .

From rendering point of view, both the terrain and the sky should have unique ids, but the current RBT doesn't handle them. The temporal solution is assigning arbitrary ids to the sky and the terrain.
This is OK for now, but I feel the sky and the terrain are something special and they should have always the same ids if possible.

So, I want to request that we assign unique ids for the sky (or space or world if it's appropriate) and the terrain as follows:
```
sky: 0
terrain: 1
```

sherm1, 2017-03-17:
#3088 is referring to physical bodies (in particular world), which are not geometry. RBTree currently knows about collision geometry, but not visual geometry. We are splitting out all geometry handling (both physical and collision) to GeometryWorld so your request might be for that? MultibodyTree will not have any geometry at all.

@SeanCurtis-TRI may have a useful comment since he is working on GeometryWorld.

kunimatsu-tri, 2017-03-17:
Thanks. Resolved by discussion on Slack. I withdraw the request.

For the record, we've got a consensus that we should **NOT** define terrain to have an unique id since terrain could have more specific semantics than just terrain, and that it's OK for the sky (space, world, empty or whatever) to have ID zero in GeometryWorld.

TristanThrush, 2017-08-04:
Would it be good to add support for the <static> tag (from the SDF format), so that the Drake SDF parser can handle an SDF for multiple objects and make some of them floating and some of them fixed?

EricCousineau-TRI, 2017-10-12:
Related: Other feature requests are in #7025 as well

\cc @avalenzu

sammy-tri, 2018-07-02:
Marking the model instance bits done based on #8986 and #9041

sherm1, 2018-12-05:
cc @tri-ltyyu

EricCousineau-TRI, 2020-04-16:
This done?

sherm1, 2020-04-16:
I think this is far enough along that we could close it and have separate issues if there is anything left to be done. What do you think @amcastro-tri ?

amcastro-tri, 2020-04-20:
agree.

closed 2020-04-20 by amcastro-tri