RobotLocomotion/drake #5019: Add GetSolution(Expression) in MathematicalProgram

hongkai-dai, 2017-02-01:
Currently `MathematicalProgram` has `GetSolution()` function for a variable. We should also support `GetSolution(Expression)`, as what spotless provides.

soonho-tri, 2017-02-02:
Does it evaluate a given `Expression` under the current solution assignment?

hongkai-dai, 2017-02-02:
> Does it evaluate a given Expression under the current solution assignment?

Yes, that is the plan.

manuelli, 2017-04-07:
Any update on this? Would be super useful feature for a planning project we are working on with Robin and Twan.

hongkai-dai, 2017-04-08:
We have `EvalBindingAtSolution`, does this solve your problem? Or you need to evaluate an expression that is not a constraint or a cost?

manuelli, 2017-04-08:
I took a look at `EvalBindingAtSolution` and I don't think it quite solves my problem since I need to evaluate something that isn't a constraint or cost. Essentially the use case is as follows. There are some variables `x y`. I make an expression `z = x + y` which appears often in constraints, costs etc. For example `AddLinearConstraint(z <= 1)`. I would like to get the value of `z` at the optimal solution.

hongkai-dai, 2017-04-08:
Currently the simplest solution is to use `Evaluate` function in `symbolic_expression.h`. You will need to construct a `symbolic::Environment`, to store the mapping between the symbolic variables, and their values at the solution. The easiest way is to do
```C++
const auto& vars = e.GetVariables();
const Eigen::VectorXd& var_val = prog.GetSolution(vars);
symbolic::Environment env;
for (int i = 0; i < vars.rows(); ++i) {
  env.insert(vars(i), var_val(i));
}
e.Evaluate(env);
```

I will wrap up this code block into `GetSolution(const Expression& e)` later in a PR. For the moment, could you try this code snippet first?

manuelli, 2017-04-08:
Ok I am on working through the Python API currently but will make a c++ example and try it out. It turns out the expression is linear, so maybe I could make just make it into a linear cost or constraint and then evaluate it.

hongkai-dai, 2019-03-25:
We have `MathematicalProgramResult::GetSolution(const symbolic::Expression& e) const` now.

closed 2019-03-25 by hongkai-dai