RobotLocomotion/drake #5022: Update Bazel CI matrix

david-german-tri, 2017-02-02:
From f2f discussion, as the Bazel build gets more comprehensive, we need more CI:

In **nightly production**, we should have the full product of all of the following binary variables on Xenial and Trusty:

* clang/gcc
* debug/release
* open-source/everything

We should also have four builds on OS X spanning debug/release and open-source/everything.

In **continuous production**, we should have the full product on Xenial, plus at least one Trusty build, probably gcc-release-everything.

In **pre-merge**, we should pick three or four Linux builds that span Trusty/Xenial and the other variables in a reasonable way.

Some of these configurations have never been tested in CI, so will need to be started as nightly, fixed if necessary, and upgraded over the next few days.

david-german-tri, 2017-02-02:
The proprietary build (SNOPT in particular) will also require upgrading to Bazel 0.4.3 per #5020.

BetsyMcPhail, 2017-02-02:
## Nightly Production
### Already Exist
linux-xenial-clang-bazel-nightly-debug
linux-xenial-clang-bazel-nightly-release
linux-clang-bazel-nightly-debug
linux-clang-bazel-nightly-release
linux-xenial-gcc-bazel-nightly-debug
linux-xenial-gcc-bazel-nightly-release
linux-gcc-bazel-nightly-debug
linux-gcc-bazel-nightly-release
mac-clang-bazel-nightly-release
### Need to Add
linux-xenial-clang-bazel-nightly-open-source-debug
linux-xenial-clang-bazel-nightly-open-source-release
linux-clang-bazel-nightly-open-source-debug
linux-clang-bazel-nightly-open-source-release
linux-xenial-gcc-bazel-nightly-open-source-debug
linux-xenial-gcc-bazel-nightly-open-source-release
linux-gcc-bazel-nightly-open-source-open-source-debug
linux-gcc-bazel-nightly-open-source-release
mac-clang-bazel-nightly-debug
mac-clang-bazel-nightly-open-source-release
mac-clang-bazel-nightly-open-source-debug

## Continuous production
### Already Exist
linux-gcc-bazel-continuous-debug
linux-gcc-bazel-continuous-release
linux-xenial-clang-bazel-continuous-debug
linux-xenial-clang-bazel-continuous-release
linux-xenial-gcc-bazel-continuous-debug
linux-xenial-gcc-bazel-continuous-release
### Need to Add
linux-xenial-clang-bazel-continuous-open-source-debug
linux-xenial-clang-bazel-continuous-open-source-release
linux-xenial-gcc-bazel-continuous-open-source-debug
linux-xenial-gcc-bazel-continuous-open-source-release

## Pre-merge
### Already Exist
linux-gcc-bazel-experimental
linux-xenial-gcc-bazel-experimental
### Need to Add
???

BetsyMcPhail, 2017-02-10:
The following nightly jobs have been added to Jenkins:

linux-xenial-clang-bazel-nightly-everything-debug
linux-xenial-clang-bazel-nightly-everything-release
linux-clang-bazel-nightly-everything-debug
linux-clang-bazel-nightly-everything-release
linux-xenial-gcc-bazel-nightly-everything-debug
linux-xenial-gcc-bazel-nightly-everything-release
linux-gcc-bazel-nightly-everything-debug
linux-gcc-bazel-nightly-everything-release

BetsyMcPhail, 2017-02-13:
The new nightly builds look good. Moving on to the Continuous builds.

The following builds have been added:
linux-xenial-clang-bazel-continuous-everything-debug
linux-xenial-clang-bazel-continuous-everything-release
linux-xenial-gcc-bazel-continuous-everything-debug
linux-xenial-gcc-bazel-continuous-everything-release

BetsyMcPhail, 2017-02-14:
bazel / everything / debug builds in both continuous and nightly are failing with a timeout.

For example: https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=388278

david-german-tri, 2017-02-14:
Thanks - we should either disable `valkyrie_balancing_test` in Debug or bump its timeout.  @siyuanfeng-tri, any thoughts about which is better?

siyuanfeng-tri, 2017-02-14:
probably bump up the time? I did medium in cmake land, what's the equivalent here?

david-german-tri, 2017-02-14:
`small` is the default, so `medium` sounds like a good plan.

https://bazel.build/versions/master/docs/test-encyclopedia.html#role-of-the-test-runner

jamiesnape, 2017-02-14:
> I did medium in cmake land, what's the equivalent here?

The test sizing should be the same for both. The CMake support was modeled on Google's test sizes also.

BetsyMcPhail, 2017-02-22:
The nightly and continuous everything builds have been promoted to the 'Production' lists in Jenkins.

BetsyMcPhail, 2017-04-13:
With the addition of 'mac-clang-bazel-nightly-everything-release' to Nightly Production, the **nightly production** and **continuous production** lists are complete as described above.

@david-german-tri, the current **pre-merge** test list is below. Let me know if you would like to see any adjustments.

**Trusty pre-merge:**
linux-clang-experimental-debug
linux-clang-experimental-memcheck-lsan
linux-gcc-bazel-experimental
linux-gcc-experimental-matlab
linux-gcc-experimental-ros

**Xenial pre-merge:**
linux-xenial-clang-experimental
linux-xenial-clang-experimental-debug
linux-xenial-clang-ninja-experimental-minimal
linux-xenial-clang-ninja-experimental-open-source
linux-xenial-cpplint-experimental
linux-xenial-gcc-bazel-experimental
linux-xenial-gcc-experimental-open-source
linux-xenial-gcc-ninja-experimental
linux-xenial-gcc-ninja-experimental-open-source-debug

david-german-tri, 2017-04-13:
cc @stonier 

Great.  My only request for the pre-merge list is to add `linux-xenial-clang-bazel-experimental-debug`.  

Around the time of the cutover, of course, we'll want to replace the CMake builds with equivalent Bazel ones.  I know we're missing the `*san` family, ROS, and MATLAB.  Anything else?

jamiesnape, 2017-04-13:
Do you still want to a separate lint, i.e. should we exclude lint-related labels from all builds except that one? I have no preference, personally.

david-german-tri, 2017-04-13:
I have no preference either.

BetsyMcPhail, 2017-04-18:
linux-xenial-clang-bazel-experimental-debug has been created and added to the pre-merge list.

BetsyMcPhail, 2017-04-18:
As all of the immediate work is complete, I am going to close this issue.

closed 2017-04-18 by BetsyMcPhail