RobotLocomotion/drake #5036: Add screenshot visualization of collision model in model.rst

liangfok, 2017-02-03:
https://reviewable.io/reviews/RobotLocomotion/drake/4997#-KbrOMmRW5JigjlYIVPl

RussTedrake, 2017-02-03:
the matlab drake visualizer had a flag that flipped from sending the visual geometry for collision geometry.  would be trivial to do that in the c++ drake visualizer, too.

Lucy-tri, 2017-03-30:
Upon reflection, I think that adding this picture goes beyond the scope of this doc a bit.  I'm sorry I have not got to it.  The document has changed, also.  I'm handing off my models work to Allison, so I'm assigning this to her. @liangfok , do you want it?  We could also just close this issue. 
Being able to display collisions in drake-visualizer would be nice, but that's a different issue.

jwnimmer-tri, 2018-06-09:
We can revisit if / when we next refresh this document.

closed 2018-06-09 by jwnimmer-tri