RobotLocomotion/drake #5037: adding cddlib to externals

hongkai-dai, 2017-02-03:
This library is available in https://www.inf.ethz.ch/personal/fukudak/cdd_home/. It is used to compute the two representations of a polyhedron, either using vertices (V representation) or halfspaces (H representation).

@rdeits uses it for IRIS, and I will use it for computing the Newton polytope in sum-of-squares optimization.

RussTedrake, 2017-02-03:
it looks like cddlib is GPL.  is this license going to get in the way?  @jwnimmer-tri ?

liangfok, 2017-02-03:
@sherm1's statement here is relevant: https://github.com/RobotLocomotion/drake/issues/3178#issuecomment-241087827

rdeits, 2017-02-03:
That's highly unfortunate. I may need to modify iris to comply or to remove cddlib.

liangfok, 2017-02-03:
Or perhaps contact the `cddlib` authors  to see if they're willing to issue a different license like MIT or BSD.

hongkai-dai, 2017-02-03:
qhull (http://www.qhull.org/) is under MIT license https://github.com/bjnortier/qhull/blob/master/LICENSE, we can potentially use it instead of cddlib.

hongkai-dai, 2017-02-03:
qhull is not under MIT license, sorry.

closed 2017-02-03 by hongkai-dai