RobotLocomotion/drake #5044: Should not require manual creation of VTP textures.

liangfok, 2017-02-03:
Currently, one must manually convert a texture to VTP format for it to show up in Drake Visualizer. We should update Drake to not require manually conversion!

liangfok, 2017-02-03:
In the past for HSRb, I used the following modified converter:

https://github.com/liangfok/oh-distro/commit/b94870367180791431ebb152f9d2ff83439b5962

liangfok, 2017-02-15:
I'm thinking about elevating the priority of this issue because it is preventing me from properly visualizing a multi-lane road in `drake-visualizer`. The screenshot below is supposed to show a 10-lane dragway:

[image]

jwnimmer-tri, 2017-02-21:
In prior discussions, we've considered using http://www.assimp.org/ for this kind of purpose.  If nothing else, maliput's obj generator could automatically produce the new asset format by calling the conversion tool.

See also #1784 where assimp was discussed.

patmarion, 2017-02-22:
do you have a screenshot from another program to show how it's supposed to look?  The colors depend both on the object material properties and on the scene's lighting model.  Here is the result I get from drake-visualizer using `assimp2vtk`.

[image]

liangfok, 2017-02-22:
Here is what I see when using MeshLab:

[image]

I actually think the yellowish version you see is OK because that's what @jadecastro's screenshot shows here:

https://github.com/RobotLocomotion/drake/pull/5169#issue-207623565

@maddog-tri, as the author of the of the `RoadGeometry`-to-Mesh converter, what should the visualization look like?

patmarion, 2017-02-22:
Note also that the mtl file sets the opacity to 0.5 for all materials, meaning the objects are translucent, so the gray background influences the color, too.

SeanCurtis-TRI, 2017-02-22:
Sounds like an error in the mtl file.

patmarion, 2017-02-22:
correction, the 2 overlay pieces have opacity=0.5, the lower layer is not translucent.

```
# GENERATED BY maliput::utility::GenerateObjFile()
#
# DON'T BE A HERO.  Do not edit by hand.

newmtl marker_paint
Ka 0.8 0.8 0.0
Kd 1.0 1.0 0.0
Ks 1.0 1.0 0.5
Ns 10.0
illum 2
d 0.5

newmtl bland_asphalt
Ka 0.1 0.1 0.1
Kd 0.2 0.2 0.2
Ks 0.3 0.3 0.3
Ns 10.0
illum 2

newmtl lane_haze
Ka 0.9 0.9 0.9
Kd 0.9 0.9 0.9
Ks 0.9 0.9 0.9
Ns 10.0
illum 2
d 0.20
```

maddog-tri, 2017-02-23:
The "drake-visualizer using `assimp2vtk`" screenshot from @patmarion looks about right; the lane swaths, stripes, and arrows are intended to be translucent overlays (on top of opaque blacktop).  I don't know why meshlab doesn't show yellow stripes/arrows --- but I stopped using meshlab to view OBJ files a while ago.  Here is a screenshot from `view3dscene`:
[image]
I am no master of the MTL lighting parameters.

maddog-tri, 2017-02-23:
And, to add my two cents:  if `drake-visualizer` is going to make any pretense of loading OBJ files, then it should deal with the MTL files, too.  If that mean that it calls `assimp2vtk` under its hood to accomplish that, great, but the user shouldn't have to know about it.

liangfok, 2017-02-23:
@patmarion, what's your recommendation on how to resolve this issue? We currently have software in Drake that's able to write an `.obj` and `.mtl`. We'd like to automate the conversion of these files into a format that `drake-visualizer` can  use. Should we call your `assimp2vtk` program to do this?

patmarion, 2017-02-24:
fyi, https://github.com/RobotLocomotion/director/pull/475

I backported the vtkOBJImporter so that it's available in director.  If you want to use it, you can add it to `drakevisualizer.py`.  I think the vtkOBJImporter implementation has room for improvement, but it seems to work ok on the files I tested.

Lucy-tri, 2017-03-08:
Does this issue really belong in the Director repo?

patmarion, 2017-03-08:
No, I don't think so.  Director has plenty of ways to draw obj and mtl files and I've commented a lot about it already.  It's up to TRI to decide which method(s) you prefer to push forward with in drake/drake-visualizer.

patmarion, 2017-03-08:
note for anyone new to this topic- please search the drake-developers slack as there has been long discussions there previously.

Lucy-tri, 2017-03-27:
I'd like to clearly state for the record that the companion texture file format for OBJ files is the MTL texture format, but MTL is not supported by drake-visualizer. This is confusing for anyone who is trying to use an OBJ file in Drake/drake-visualizer. https://en.wikipedia.org/wiki/Wavefront_.obj_file

liangfok, 2017-04-13:
I'm going to reopen this issue until after I bump Director's git submodule SHA and update Drake's documentation.

liangfok, 2017-04-16:
This issue was resolved by #5832.

closed 2017-04-16 by liangfok