RobotLocomotion/drake #5048: Add Symbolic::Polynomial class

hongkai-dai, 2017-02-03:
This class should store a map `unordered_map<Monomial::ID, double>` monomial_to_coeff_map_. It should support following functions

1. Return the degree of the polynomial.
2. Return the degree of the polynomial for a subset of variables.
3. Return the Newton polytope.
4. Return the set of variables in the polynomial.

hongkai-dai, 2017-02-04:
The current design on `Monomial` is that we have an internal class to represent a monomial, and a function exposed to the user, that can construct a monomial, and the constructed monomial is a `symbolic::Expression` object, the code is like the following
```C++
namespace internal {
  class Monomial {
   private:
    map<Variable::Id, int> powers_; // stores the map from a variable to its exponent.
  }
}  // namespace internal

Expression Monomial(const std::unordered_map<Variable, int, hash_value<Variable>>& map_var_to_expnt);
```
The problem with this design is that `Monomial` class is not exposed to the user. The user only gets a `symbolic::Expression` object. If we want to extract the information on the exponent, we need to use the functions for `symbolic::Expression` to find the exponent. On the other hand, we actually store the `powers_` in the `internal::Monomial` class, but this information is thrown away.

One solution is to expose `Monomial` class to the user, and move it out of the `internal` namespace. We do want to use monomial like a symbolic expression. To this end we make `Monomial` *has a* symbolic expression. The class `Monomial` look like
```C++
namespace symbolic {
class Monomial {
 public:
  Monomial(const std::unordered_map<Variable, int, hash_value<Variable>>& map_var_to_expnt);

  ToExpression() const { return e_; }

 private:
  Expression e_;
  int total_degree_;
  map<Variable::Id, int> powers_;
};

template<typename _CoefficientType = double>
class Polynomial {
 public:
  Polynomial(const map<Monomial, _CoefficientType>& monomial_to_coeff_map);

  ToExpression() const { return e_; }

 private:
  Expression e_;
  map<Monomial, _CoefficientType> monomial_to_coeff_map_;
}
}
```

**Question**
1. Why do we want to use `Polynomial` as a symbolic expression? My motivation is to use `symbolic::Variable` as the variable type in `Polynomial`. Do we have other reasons? The existing `Polynomial` class uses `unsigned int` as the  variable type https://github.com/RobotLocomotion/drake/blob/master/drake/common/polynomial.h#L45
2. Do we still need to make `Polynomial` as a templated class? It can only support coefficient type being either double or `symbolic::Variable`.

hongkai-dai, 2017-02-06:
@RussTedrake and @soonho-tri, what do you think?

RussTedrake, 2017-02-06:
I'm a little confused.  My understanding is the polynomial class in drake/common/polynomial.h was originally intended to serve as a first-class citizen scalar type in drake (along with double, autodiff, ...).  But that we have now decided that there are only to be three first-class scalar types in drake (double, autodiffxd, and symbolic expression).  The system classes only need to support these three, and can do explicit template instantiations for them.  Note that autodiffxd still needs to be updated to the chunking version.  Is that all consistent with your question above?

So now the real question is -- somebody wants to write an algorithm specific to polynomials.  e.g. generating a vector of monomials, or computing a newton polytope.  what class do they do this against?  do we have a "ToPolynomial" method in symbolic expression that kicks out to an independent, special purpose Polynomial class?  Or do we write operations directly against the symbolic expression (Polynomial isa symbolic::Expression).  I thought we had nominally chosen that Polynomial isa symbolic::Expression.  

So I think what would be useful, and what you've started here, is to list the methods that this Polynomial class needs to have above and beyond the more generic symbolic::Expression?   generating monomials, newton polytope (which requires accessing the powers, ...), etc.  

Sorry if I'm being slow.  Just don't want to talk past each other.

hongkai-dai, 2017-02-06:
@RussTedrake 
> My understanding is the polynomial class in drake/common/polynomial.h was originally intended to serve as a first-class citizen scalar type in drake (along with double, autodiff, ...).

That is also my understanding.

> So now the real question is -- somebody wants to write an algorithm specific to polynomials. e.g. generating a vector of monomials, or computing a newton polytope. what class do they do this [...]

That is where I am not so sure, whether we want to create a special Polynomial class. I think this is related to how we represent the polynomial class internally, if we were to create a Polynomial class. I can think of two internal representations

1. A map `map<Monomial, double> monomial_to_coeff`. If we want to use this representation, then what we have in the `symbolic::Expression` class is sufficient, that we have a `ExpressionAdd::map<Expression, double> expr_to_coeff_map`, which I believe is sufficient, although it might not sort the key `Expression` as the order we want.
2. As what we have in spotless, we use three quantities to represent a polynomial: coefficients, powers and variables, namely the returned result of `decomp` function in spotless. This representation makes it easier to compute the degree, the Newton polytope. Note that if we use `symbolic::Expression` to represent a polynomial, we can still get these information by doing some computation, but we do NOT store these information. If we choose this representation, I would rather create a special purpose Polynomial class.

The list of methods that I want to implement in this Polynomial class beyond symbolic::Expression is

1. Computing the degrees of a polynomial, w.r.t some variables.
2. Computing the Newton polytopes. 
3. Generating a polynomial with some monomials.

RussTedrake, 2017-02-06:
The problem with having a separate class is that we have two sets of methods to maintain for symbolic operations.  Things like simplification are in symbolic::Expression, but not yet in Polynomial, etc.  

All of those methods seem reasonable to add to a class that derives from symbolic::Expression.  But @soonho-tri knows best.

hongkai-dai, 2017-02-06:
I get the first version of computing degrees of polynomial from symbolic::Expression class. I think I do not need to create a separate special purpose Polynomial class.

soonho-tri, 2017-02-07:
>  I think I do not need to create a separate special purpose Polynomial class.

I agree.

Here are my thoughts:

 - Please note that we added `Polynomial ToPolynomial(const Expression&)` because `Expression` doesn't have `Expand` method while `Polynomial` supports it at construction. We wanted to take a shortcut by using `Polynomial`. I'll implement `Expression::Expand` method.

 - Q: Do we want to expand `p1 * p2` at construction where `p1` and `p2` are `symbolic::Expression`s representing polynomials (that is, is_polynomial() returns true for `p1` and `p2`)?

 - @hongkai-dai , I can also implement `T DecomposePolynomial(const Expression& p)` method which assumes that `p.is_polynomial() == true`. We can discuss its return type `T`. We can take care of the ordering issue here (by taking a variable ordering as a template argument?).

RussTedrake, 2017-02-07:
If I understand “expand”, then whether to expand on construction or not seems like an implementation detail (albeit an important one).  But presumably we do need to be able to get correct answers to all of the polynomial questions, either by recursing through the symbolic::Expressions or by having the expanded polynomial available locally?

soonho-tri, 2017-02-07:
> If I understand “expand”, then whether to expand on construction or not seems like an implementation detail (albeit an important one).

Correct.

> But presumably we do need to be able to get correct answers to all of the polynomial questions, either by recursing through the symbolic::Expressions or by having the expanded polynomial available locally?

That is correct as well.

soonho-tri, 2017-02-07:
@hongkai-dai and I just had a video-chat. Here are outcomes from the discussion:

 - We agreed that it's better to use `symbolic::Expression` to represent a polynomial. We won't have a separate class for a polynomial.
 - I'll implement `Expression::Expand` method but this will not be called by the time an expression is constructed. For example, `e1 * e2` won't be expanded, neither `pow(e1, e2)`.
 - I'll implement more simplification regarding division(`/`). Currently there are some missing cases.
 - @hongkai-dai is already working on `DecomposePolynomial`. The current signature of this function is:
    ```c++
    void Expression::DecomposePolynomial(Eigen::ArrayXd* coeff, Eigen::ArrayXXd* power,
                                         std::vector<Variable>* vars);
    ```
 - I'll add visitor pattern to the `Expression` class so that we can easily write a non-member function processing a symbolic expression.

PR #5073 (2017-02-07): Polynomial degree
Partially resolves #5048

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5073)
closed 2017-02-07 by hongkai-dai