RobotLocomotion/drake #5058: (legitimate) compilation error on libbot2

RussTedrake, 2017-02-05:
not sure how we never hit this before
```
Ld /Users/russt/drake-distro/build/externals/libbot/bot2-frames/lib/Debug/libbot2-frames.2.dylib normal x86_64
    cd /Users/russt/drake-distro/externals/libbot/bot2-frames
    export MACOSX_DEPLOYMENT_TARGET=10.12
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -arch x86_64 -dynamiclib -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -L/Users/russt/drake-distro/build/externals/libbot/bot2-frames/lib/Debug -L/Users/russt/drake-distro/build/externals/libbot/bot2-frames/lib -L/Users/russt/drake-distro/build/externals/libbot/bot2-frames/lib/Debug/Debug -L/Users/russt/drake-distro/build/install/lib/Debug -L/Users/russt/drake-distro/build/install/lib -F/Users/russt/drake-distro/build/externals/libbot/bot2-frames/lib/Debug -filelist /Users/russt/drake-distro/build/externals/libbot/bot2-frames/src/bot2-frames.build/Debug/bot2-frames.build/Objects-normal/x86_64/bot2-frames.2.LinkFileList -install_name /Users/russt/drake-distro/build/externals/libbot/bot2-frames/lib/Debug/libbot2-frames.2.dylib -Xlinker -rpath -Xlinker /Users/russt/drake-distro/build/externals/libbot/bot2-frames/lib -Xlinker -rpath -Xlinker /Users/russt/drake-distro/build/install/lib -mmacosx-version-min=10.12 -Xlinker -object_path_lto -Xlinker /Users/russt/drake-distro/build/externals/libbot/bot2-frames/src/bot2-frames.build/Debug/bot2-frames.build/Objects-normal/x86_64/bot2-frames.2_lto.o -Xlinker -no_deduplicate -dynamiclib -Wl,-headerpad_max_install_names /Users/russt/drake-distro/build/externals/libbot/bot2-frames/lib/Debug/liblcmtypes_bot2-frames.a -L/Users/russt/drake-distro/build/install/lib -L/usr/local/Cellar/glib/2.50.2/lib -L/usr/local/opt/gettext/lib -lbot2-param-client -lbot2-core -lm -llcmtypes_bot2-core -lgthread-2.0 -llcmtypes_bot2-param -llcm -lglib-2.0 -lintl -single_module -compatibility_version 2.0.0 -Xlinker -dependency_info -Xlinker /Users/russt/drake-distro/build/externals/libbot/bot2-frames/src/bot2-frames.build/Debug/bot2-frames.build/Objects-normal/x86_64/bot2-frames.2_dependency_info.dat -o /Users/russt/drake-distro/build/externals/libbot/bot2-frames/lib/Debug/libbot2-frames.2.dylib
ld: warning: directory not found for option '-L/Users/russt/drake-distro/build/externals/libbot/bot2-frames/lib/Debug/Debug'
ld: warning: directory not found for option '-L/Users/russt/drake-distro/build/install/lib/Debug'
Undefined symbols for architecture x86_64:
  "_bot_core_pose_t_subscribe", referenced from:
      _bot_frames_new in bot_frames.o
  "_bot_core_pose_t_unsubscribe", referenced from:
      _frame_handle_destroy in bot_frames.o
  "_bot_core_rigid_transform_t_subscribe", referenced from:
      _bot_frames_new in bot_frames.o
  "_bot_core_rigid_transform_t_unsubscribe", referenced from:
      _frame_handle_destroy in bot_frames.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

bot2-frames is clearly depending on these messages:
https://github.com/RobotLocomotion/libbot2/blob/master/bot2-frames/src/bot_frames.c#L25

but they are defined in bot_core_lcmtypes (which depends on libbot, and therefore is build after libbot).  Seems like a clear circular dependency to me.  

The relevant repos haven't changed very recently.  I can understand how I might have had a build artifact on my machine that didn't trigger until now, but how can the build servers be letting this through?

/cc @patmarion

patmarion, 2017-02-05:
not sure about your error, but I can tell you that libbot does not depend on bot_core_lcmtypes, and bot_core_lcmtypes does not depend on libbot.

RussTedrake, 2017-02-05:
not according to https://github.com/RobotLocomotion/drake/blob/master/CMakeLists.txt#L230 and the link in my post (in which libbot clearly depends on the type which is defined in bot_core_lcmtypes?)

patmarion, 2017-02-05:
Libbot has all the lcmtypes it needs to compile by itself.  bot_core_lcmtypes is a standalone repo of lcmtypes and duplicates some lcmtypes from libbot.

RussTedrake, 2017-02-05:
ok.  confirmed that libbot2 has a copy of those messages, too.  will do some more digging.

RussTedrake, 2017-02-05:
ok.  i  think the problem is simply what is pointed out in the warnings: there is an extra `Debug` that is being appended to the library path.  `/Users/russt/drake-distro/build/externals/libbot/bot2-frames/lib/Debug/Debug` does not exist, but `'-L/Users/russt/drake-distro/build/externals/libbot/bot2-frames/lib/Debug` does exist (and contains the missing library).

Sigh.  Probably xcode-specific.

RussTedrake, 2017-02-05:
Thanks @patmarion

mwoehlke-kitware, 2017-02-06:
> bot_core_lcmtypes does not depend on libbot

It ["does"](https://github.com/RobotLocomotion/drake/blob/master/CMakeLists.txt#L231 ); per the comment, that is just for ordering so that they don't both try to concurrently create an LCM types library with the same name. You might even say it's an "anti-dependency" :slightly_smiling_face:.

>  Probably xcode-specific

Given this appears related to multi-config generators, that's probably the case... (Well, specific to multi-config generators, anyway.)

patmarion, 2017-02-06:
Yup, which I have documented here (on the old repo url): https://github.com/RobotLocomotion/libbot/issues/20

Would love to see that issue resolved in some way, it's an ugly one.

david-german-tri, 2017-04-18:
Closing as wontfix since Russ isn't using xcode anymore.

closed 2017-04-18 by david-german-tri