RobotLocomotion/drake #5071: add open-source QP solvers

hongkai-dai, 2017-02-06:
per discussion with @edrumwri , we should consider moby, quadprog++, etc.

siyuanfeng-tri, 2017-02-07:
i have used an eigenfied version of quadprog++ in the past.

hongkai-dai, 2018-02-01:
Lift the priority of this issue, as Siyuan mentions that we want to solve QP problem at high rate on the robot. Mosek and Gurobi has the license issue (they need to talk to the license server, thus limiting its running speed). SCS is unfortunately inaccurate for this problem. So currently on the robot the code calls SNOPT, which can fail occasionally.

hongkai-dai, 2018-02-01:
@jamiesnape and @jwnimmer-tri , can we build quadprog++ (https://github.com/liuq/QuadProgpp) in drake through bazel?

jwnimmer-tri, 2018-02-01:
Yes, it looks trivial.  It would be similar to how we build the `fmt` library (see `tools/workspace/fmt/**` as called by `tools/workspace/default.bzl`), for example.

jamiesnape, 2018-02-01:
@hongkai-dai Create an issue an assign it to me if you are not comfortable. Since it is a private dependency, it simplifies things even more.

hongkai-dai, 2018-02-01:
Thanks! I can create a bazel build file for this.

RussTedrake, 2018-02-02:
cc @frankpermenter

arocchi, 2018-02-05:
You guys should take a look at qpOases

hongkai-dai, 2018-02-05:
Thanks @arocchi , qpOASES looks great!

Capri2014, 2018-02-05:
qpOASES is relatively slow through..

frankpermenter, 2018-02-08:
A new solver with impressive benchmarks is OSQP.  See:

http://osqp.readthedocs.io/en/latest/
https://github.com/oxfordcontrol/osqp_benchmarks

PR #8029 (2018-02-20): Add OSQP
Resolves #5071 

@rdeits .

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8029)
closed 2018-02-20 by hongkai-dai