RobotLocomotion/drake #5072: Promote Parameters from LeafContext to Context

david-german-tri, 2017-02-06:
Contrary to my original belief, there are a few reasons we might want to reason about Parameters independent of a particular LeafSystem's semantics.  One of these is to measure symbolic sparsity.

I vaguely recall that @jadecastro had some other reasons.

jadecastro, 2017-02-07:
Yeah, it fits well in the realm of parameter-varying systems, where the parameters for a given system are allowed to evolve according to some external process. A simple example of this is linearization of some nonlinear system (where the equilibria can vary with time and are not described by the linearized dynamics).

RussTedrake, 2017-02-07:
Parameter estimation seems like another clear example.  There are generic tools for this (e.g. take the gradient of the dynamics with respect to the parameters, and update in that direction) that need not understand LeafSystem semantics to be used.

PR #5467 (2017-03-14): Move Parameters from LeafContext to Context.
Fixes #5072.

@jadecastro for feature review

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5467)
commit 561ccd11 (2017-03-14): Move Parameters from LeafContext to Context.

Fixes #5072.
closed 2017-03-14 by david-german-tri