RobotLocomotion/drake #5078: Questions about linearApproxHJB.m

jefesaurus, 2017-02-07:
@RussTedrake Looks like you added this a few years ago: https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/systems/dev/linearApproxHJB.m -- Did you ever get it working? I don't  understand most of the math, but from the comments it seems like it is setting up a equations between mesh points, sample points and interpolation weights in the first loop and then the lower loop is some gradient descent to approximate the `alpha` values(which I believe are values defining the mesh)? Is that right, and did you ever get the `alpha` values to stabilize for the double integrator? All I see is like this with no sign of convergence:

[image]

RussTedrake, 2017-02-07:
Yes - I believe i got this in and working (probably the night before a lecture), but it's likely that something else has broken it since it wasn't covered by CI.

This is the natural implementation of "fitted value iteration" (motivated by td learning with function approximators) for continuous systems as described in the header.

Moving the dynamics programming examples like this to C++ is on my list and i don't see any blockers.  Could potentially help step you through it if you're willing to contribute to that port.

jefesaurus, 2017-02-08:
Okay, I just built the repo from 2014, and found the same results as I do currently. I think I'll try the MDP approximation to continuous-state value iteration update you suggest in 9.4.2 of http://underactuated.csail.mit.edu/underactuated.html?chapter=9

Seems a little bit simpler to try out in C++

closed 2017-05-01 by RussTedrake