RobotLocomotion/drake #5081: Unable to start drake-visualizer on OSX

liangfok, 2017-02-07:
I'm not sure when this started, but I just noticed this morning that I'm unable to start `drake-visualizer` on my OSX machine.

# The Error

```
$ cd drake-distro
$ ./build/externals/director/bin/drake-visualizer
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/liang/dev/drake-distro-6/build/externals/director/lib/python2.7/dist-packages/director/drakevisualizerapp.py", line 3, in <module>
    from director import mainwindowapp
  File "/Users/liang/dev/drake-distro-6/build/externals/director/lib/python2.7/dist-packages/director/mainwindowapp.py", line 3, in <module>
    from director import consoleapp
  File "/Users/liang/dev/drake-distro-6/build/externals/director/lib/python2.7/dist-packages/director/consoleapp.py", line 6, in <module>
    from director import applogic
  File "/Users/liang/dev/drake-distro-6/build/externals/director/lib/python2.7/dist-packages/director/applogic.py", line 5, in <module>
    import director.vtkAll as vtk
  File "/Users/liang/dev/drake-distro-6/build/externals/director/lib/python2.7/dist-packages/director/vtkAll.py", line 2, in <module>
    from vtkDRCFiltersPython import *
ImportError: dlopen(/Users/liang/dev/drake-distro-6/build/externals/director/lib/python2.7/dist-packages/director/vtkDRCFiltersPython.so, 2): Library not loaded: liblcmtypes_bot2-core.dylib
  Referenced from: /Users/liang/dev/drake-distro-6/build/install/lib/libbot2-frames.2.dylib
  Reason: image not found
```

The missing library exists:

```
$ find . -name "liblcmtypes_bot2-core.dylib"
./build/externals/bot_core_lcmtypes/lcmtypes/liblcmtypes_bot2-core.dylib
./build/install/lib/liblcmtypes_bot2-core.dylib
```
Thus, I wonder if it's an `rpath` issue.

liangfok, 2017-02-07:
@david-german-tri I assigned this to you in the hopes that you would know who to best reassign it to.

david-german-tri, 2017-02-07:
Please start by root-causing it yourself using `otool -L`, print statements in the CMake, and so on.

liangfok, 2017-02-08:
I did a clean build of Drake using SHA aab61e89c25d95171d9c14cca7a2745e32f70a94 and observed slightly different behavior. I'm able to start Drake Visualizer using the following commands:

```
$ cd drake-distro/build
$ ./install/bin/drake-visualizer
```

However, I'm unable to start Drake Visualizer using the following commands:

```
$ cd drake-distro/build
$ ./externals/director/bin/drake-visualizer
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/liang/dev/drake-distro-2/build/externals/director/lib/python2.7/dist-packages/director/drakevisualizerapp.py", line 31, in main
    fields = fact.construct(options, app=app.app, view=app.view)
  File "/Users/liang/dev/drake-distro-2/build/externals/director/lib/python2.7/dist-packages/director/componentgraph.py", line 149, in construct
    newFields = initFunction(inputFields)
  File "/Users/liang/dev/drake-distro-2/build/externals/director/lib/python2.7/dist-packages/director/mainwindowapp.py", line 437, in initDrakeVisualizer
    from director import drakevisualizer
  File "/Users/liang/dev/drake-distro-2/build/externals/director/lib/python2.7/dist-packages/director/drakevisualizer.py", line 17, in <module>
    import bot_core as lcmbot
ImportError: No module named bot_core
```

liangfok, 2017-02-16:
I tested this again using a clean build of 8ce03b9ba016891672d232893142b4df38bba407 and the status is the same. On OSX, `drake-visualizer` can be started using:

```
cd drake-distro/build
./install/bin/drake-visualizer
```

but **cannot** be started using:

```
cd drake-distro/build
./externals/director/bin/drake-visualizer
```

due to the same error mentioned above. However, since the installed version works, I believe it is sufficient to close this issue.

closed 2017-02-16 by liangfok