RobotLocomotion/drake #5085: Unable to View Body Frames in Drake Visualizer

liangfok, 2017-02-07:
# Problem Definition

We are currently unable to view body frames using DrakeVisualizer, as shown below:

[image]

In the [past](https://github.com/RobotLocomotion/drake/issues/1714#issuecomment-193809869), we were able to view them, as shown below:

[image]

It would be nice to restore that functionality.

liangfok, 2017-02-07:
@patmarion I assigned this to you in the hopes that you would know what happened. Please free to re-assign if you don't have time to restore this functionality.

patmarion, 2017-02-07:
frame visualization was removed with this PR:  https://github.com/RobotLocomotion/director/pull/362

it was removed because there was a slowdown associated with drawing 1000s of frames at high rate.  the feature could be added back and implemented in a way to avoid the slowdown.  I'd be happy to advise if you want to work on it.

liangfok, 2017-02-07:
@patmarion Thanks for the quick response! This feature is not urgently needed right now. I'll close this issue and re-open it once it's urgently needed, at which time we'll probably come looking for your advice.

closed 2017-02-07 by liangfok