RobotLocomotion/drake #5086: Incorrect detection of non-linear constraint

rdeits, 2017-02-07:
I'm trying to add a constraint to a MathematicalProgram, but I'm getting an error when checking the expression for linearity: 

```python
p = mp.MathematicalProgram()
q = p.NewContinuousVariables(2)
p.AddLinearConstraint(q[1] == q[0] + 0.5 * (q[0] + q[1]))
```

gives:

```
Expression ( - x(0) + x(1) - 0.5 * (x(0) + x(1))) is non-linear.
```

even though the expression is actually linear.

hongkai-dai, 2017-02-07:
Thanks, I will work on this.

rdeits, 2017-02-07:
thanks! :smile:

hongkai-dai, 2017-02-07:
@soonho-tri is working on `Expand()` that expands a symbolic expression. I will try to provide a quick solution, to recursively extract the linear coefficients and constant from a symbolic expression. I think eventually we will just migrate to using `Expand()`.

hongkai-dai, 2017-02-07:
A quick fix is not to use parenthesis in your expression. I think
```C++
p.AddLinearConstraint(q[1] == q[0] + 0.5 * q[0] + 0.5 *q[1])
```
should work

rdeits, 2017-02-07:
Ok, thanks. Removing the parentheses does work.

closed 2017-03-06 by hongkai-dai