RobotLocomotion/drake #5091: Visualize contact related information

siyuanfeng-tri, 2017-02-07:
Both the dynamics and robot people want to visualize what's going on with contacts, in particular, we want to see the contact positions + normal to start. We also want to be able to see the actual force being applied. 
It would be extra nice to have the interface general enough to support similar information from different sources (sensors, controller output, etc) other than RBP. 

@patmarion @liangfok for advice. 
cc @SeanCurtis-TRI @sammy-tri

patmarion, 2017-02-07:
cc @rdeits

liangfok, 2017-02-07:
This is related to https://github.com/RobotLocomotion/drake/issues/1908

liangfok, 2017-02-07:
I wonder if GeometryWorld would play a role in terms of aggregating the contact state among everything in the simulation and providing this information to an LCM publisher that sends it to Drake Visualizer.

jwnimmer-tri, 2017-02-07:
My general advice would be to publish the detail outputs you want directly as a specific message for that port (e.g., the `contact_results_output_port` port of RBP), and then add code in Director to render that message in the way you want to see it.  I advise against trying to generalize a solution beyond that simple pattern.  To my estimation, there isn't that much code that would be shared through genericizing (and if there is we just rule-of-3 it later), and the flexibility of keeping `contact_results` separate from `depth_sensor` should make it easier to maintain over time.

siyuanfeng-tri, 2017-02-07:
fair enough.

patmarion, 2017-02-07:
note, director also has general purpose lcm messages for drawing.  So there are two strategies, and it sounds like @jwnimmer-tri is recommending the first one if I'm not mistaken.

1. in drake c++, publish custom data messages.  in drake-visualizer, add custom lcm subscribers to to receive and implement drawing logic

2. in drake c++, interpret data and publish drawing messages.  nothing to be added to drake-visualizer

jwnimmer-tri, 2017-02-07:
I have also done option 2 in the past for basic geometry (e.g., "draw the planned trajectory's projection onto the xy plane as a blue polyline") -- but for rich information that will be durable (which seems like contact and sensors fits into) I highly recommend making the visual representation choices in a high-level language like Python in Director via option 1.  It will offer a rich vocabulary of both visual primitives and programmer abstractions, versus something like lcmgl or work-alike to it, where you have to program a rendering engine over one-way LCM messages with almost no lever for abstractions -- and that gets old, fast.

siyuanfeng-tri, 2017-02-07:
just a random concern. 
if i publish contact related stuff fin a separate lcm channel, presumable, it can get out of sync with the state that's being rendered? 
if i want to keep them synced, what's the best way to do it? have them in one message? the contact message also contans q, qd, and overwrites the other?

liangfok, 2017-02-07:
Since the underlying transport is UDP, I believe they can can out of sync or the packet could get lost, but the probability is low.

What about including a time stamp in the message? Would that allow Director to keep the visualization
 in sync?

jwnimmer-tri, 2017-02-07:
@liangfok UDP loss is a non-issue.

@siyuanfeng-tri Well, honestly I think the long-term answer is to use protobufs and then Magic Happens (I can explain more but not in this thread, at least not right now.)

For now though, is it truly a problem to have the poses and contact visualizations unsynced at the timescale that can actually occur in practice?  If not, then you're done.  If yes, then on the Director side the timestamps in the messages need to be accounted for.

siyuanfeng-tri, 2017-02-07:
@jwnimmer-tri most likely now, but when you pause at the wrong moment and zoom in so much, and the contact ball is hanging off some surface, you start wondering what's really going.

jwnimmer-tri, 2017-02-07:
In that case, pause could be made in-band timestamp aware pretty easily, I think?

david-german-tri, 2017-02-07:
> the contact message also contans q, qd, and overwrites the other?

Since no one else has explicitly said it: please don't duplicate information across messages.  Duplication is even more painful in protocols than it is in logic.  :+1: for all suggestions that amount to "use the timestamps as much as you need to".

jwnimmer-tri, 2017-02-08:
To be clear, my posts above are my opinions from experience.  There are plenty of good ways to get the job done, and I don't mean to object to a motivated developer solving it one way or another; my goal is only to offer help and advice.

closed 2017-02-10 by siyuanfeng-tri