RobotLocomotion/drake #5092: controlled_iiwa_world_demo does not knock down all targets

Lucy-tri, 2017-02-07:
The controlled_iiwa_world_demo has comments indicating that it knocks down all 3 cylinders on it's table, however, when I built using Bazel on Ubuntu, it only knocked down the center cylinder. 

Consider creating a test that each cylinder is in fact knocked over (perhaps check their final resting place at the end of the simulation run).

naveenoid, 2017-02-22:
I will sort this out as soon as my bazel build issues settle (hopefully in next couple of days). Your suggestion for a test on cylinder state is worth implementing.

naveenoid, 2017-03-02:
Ok since Siyuans controller update #5339 it does seem to be the case that all targets are knocked over. I can add a test anyhow.

sammy-tri, 2019-03-04:
The demo has been deleted, along with the associated comment.  The functionality does not appear to have been reproduced anywhere else.  Closing as there's nothing left to do.

closed 2019-03-04 by sammy-tri