RobotLocomotion/drake #5096: Rename CompareToClone() methods to be IsExactlyEqualTo().

liangfok, 2017-02-08:
See: https://reviewable.io/reviews/robotlocomotion/drake/4958#-KcPp5CPEjY7xHrKgKPL

amcastro-tri, 2017-02-08:
Is it really worth spending your energy on this @liangfok? remember this is just part of a sprint to get some partial support for RigidBodyPlant templated on AutoDiffScalar.

liangfok, 2017-02-08:
@david-german-tri asked that it be done. His explanation is given in the link in this issue's description. I've copy/pasted it below:

> If not, please evict them to a test-only library and rephrase them as free functions. Test-only logic in production classes is an antipattern, because it makes it harder for future readers to tell what the actual code under test is, and because it invites accidental production usage of code that was never intended to be in production.

That being said, I will set the priority of this issue to be backlog since `RigidBodyPlant` will soon be supplanted by `MultiBodyWorld` (or was it called `MultiBodyPlant`?) in the near future.

jwnimmer-tri, 2017-02-08:
If we are going to invest in Clone on RBT in order to support actual current work, then we need to do it correctly so as to sustain the work.  We can't ignore or indefinitely delay fixing defects.  I remain unconvinced that RBT Clone is a worth project, but others think it is, so let's do it right!

liangfok, 2017-02-08:
Alright. I'll upgrade the priority a bit.

PR #5189 (2017-02-16): Moves CompareToClone() methods into test-only code.
Resolves #5096.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5189)
closed 2017-02-16 by liangfok