RobotLocomotion/drake #5098: Improve CDash reporting of Bazel-run unit tests

jwnimmer-tri, 2017-02-08:
When CI runs Bazel tests, it posts the results to the dashboard in such a way that the usual mechanism for clicking through directly to test failure outputs is not specific (it usually just gives the entire log, if I recall correctly).  It would be nice to have only the specific failing tests' output.

jwnimmer-tri, 2017-02-08:
Possibly there is a good integration seam using https://github.com/bazelbuild/dash, where the Bazel dash APIs could be terminated into something that emits CDash input files or even a direct http-to-http translation (if I understand the tools correctly).

david-german-tri, 2017-02-08:
@jamiesnape and I also discussed this f2f today.  It isn't top priority but would definitely make everyone's life a little easier.

jwnimmer-tri, 2017-02-08:
Agreed on both points.

I also like it because it would also be a nice give-back to the open-source community, assuming we can upstream it somehow (and I would support doing so).

jamiesnape, 2017-11-15:
Fixed by https://github.com/RobotLocomotion/drake-ci/pull/87.

closed 2017-11-15 by jamiesnape