RobotLocomotion/drake #5105: Bazel errors do not appear in CDash

Lucy-tri, 2017-02-08:
This build has errors:
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-gcc-bazel-continuous-debug/381/
The problem is that if you click on "View in CDash" then click on the build errors, 4 errors are shown, but none of them are actually errors (they are warnings) and none of them point to the true error which you can see in the Console Output:
`//drake/examples/schunk_wsg:schunk_wsg_lift_test                        TIMEOUT in 60.1s`

I'm told this is due to this being a Bazel build. 
@jamiesnape said in Slack: 
> We should probably suppress those non-errors. No reason CDash could not work for Bazel otherwise.

jamiesnape, 2017-02-08:
They are test failures. CDash previously only displayed build errors.

jamiesnape, 2017-02-08:
See https://github.com/RobotLocomotion/drake-ci/commit/05698bcc2225f5a2baaf958a39b8c66f183ee638.

jamiesnape, 2017-02-08:
https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=376028

david-german-tri, 2017-02-08:
Nice!

closed 2017-02-08 by david-german-tri