RobotLocomotion/drake #5112: Jenkins' build links assume RobotLocomotion/drake when it's drake-ci.

Lucy-tri, 2017-02-09:
If a build fails, Jenkins has a page describing the changes that triggered that build, with links to the Github commit. For example: 
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-gcc-bazel-continuous-debug/388/

If the commit was in drake-ci, then the link won't work because the link was constructed assuming the commit was to  RobotLocomotion/drake. The above link shows the problem.

jamiesnape, 2017-04-14:
This is probably not fixable since it would involve modifying the behavior of several different plugins to support the deprecated multiple SCMs plugin.

EricCousineau-TRI, 2018-02-08:
@BetsyMcPhail (and others subscribed to this thread) Is this still an issue? I have not encountered this as a defect, so I'd be fine closing this as Not a Problem unless it arises again.

jamiesnape, 2018-02-08:
It is an issue that is not fixable, and I would say not a huge problem.

closed 2018-02-08 by jamiesnape