RobotLocomotion/drake #5116: Maliput design doc

jwnimmer-tri, 2017-02-09:
Update and publish the Maliput design document into the Drake source tree.

jwnimmer-tri, 2017-03-17:
@maddog-tri A couple times per week someone asks me for this.  Would it be sensible to get a current draft into master, even if there are further refinements to do?

maddog-tri, 2017-03-21:
Aye.  Where should it go?  

The end result is a PDF.  The source is a `.org` (text in org-mode) and `.svg` (figures in inkscape), which go through intermediate LaTeX and PDF pupation respectively, and finally emerge as the PDF.  The raw source files should definitely get bundled together in a directory; the end PDF should probably just sit in that directory as well (since that is where it gets built).

jwnimmer-tri, 2017-03-21:
Maybe `drake/automotive/maliput/doc/*`?

For now, I would be satisfied with just the `org` and `svg` bits going into git ASAP, so that we can collaboratively iterate on any ongoing improvements, and at least give people something to read for now, even if the pretty PDF isn't rendered yet.

Also, I think it's unlikely to survive in `org` format for very long, and someone on our team should soon take the patch pumpkin on it and reformat it into Doxygen.  We should only allow documentation that is written in a widely-known source format into Drake, for the same reasons we don't allow Lua or Algol code -- so that developers only need to learn a limited set of tools to be effective.  For now, getting the already-written prose into git at all trumps that need, but soon enough we should make it collaboratively maintainable by changing its markup technology.

maddog-tri, 2017-03-21:
Sounds good.

The pretty PDF is already rendered (it's the svg bits that are sorely lacking); org-mode makes that trivial if you are running emacs.

I expected the org-format to change --- it's just my go-to format for dashing stuff out in emacs (because it has nice on-screen editing structure, and it handles LaTeX math transparently enough to be both quick and useful).  However, I doubt Lua or Algol would be an appropriate source format for this particular document in the context of this project.  We could probably squeeze it all into Doxygen (which support LaTeX math, right?).  Or, worst-(best-?)case, just make it good ol' LaTeX.

jwnimmer-tri, 2017-03-21:
> The pretty PDF is already rendered...

Right, but we shouldn't commit that into `git`.

> Doxygen (which support LaTeX math, right?) ...

Yes.

caguero, 2017-06-09:
Would you consider to move the design document to a Google doc? This might improve the collaboration/discussion with multiple people. You can always take a snapshots of the last design and push it into the repository if you want.

maddog-tri, 2017-06-09:
Not really --- the considerable amount of LaTeX math in the original document would be butchered by a round-trip into a Google doc (and... once I finally get some figures/diagrams into the doc, they would suffer, too).  Also, I doubt there would be much to be gained by "google doc comments" versus just posting comments here.  

Hmm... it occurs to me that there is no record here of the current doc being committed to the repo.  There --- I just added a cross-reference in the original PR, #5603.

caguero, 2017-06-15:
Fair points, thanks.