RobotLocomotion/drake #5132: Create a "Dynamics Team Style Guide" For Simulation Parameters

liangfok, 2017-02-10:
The motivation for this issue is based on @amcastro-tri's comment here:

https://reviewable.io/reviews/RobotLocomotion/drake/4958#-KcTF7cRY_Rc5LezcAMX

Key quote:

> For simulation choosing numbers at random is as bad practice as not following the style guide for C++ (actually worst in my opinion).

I think formally documenting rules that capture hard-earned wisdom like the above will be of great benefit to anyone who wants to use Drake.

jwnimmer-tri, 2017-02-10:
For the record, I don't see in that code how "not following the style guide" comes into play.

liangfok, 2017-02-11:
I believe @amcastro-tri was just trying to make an analogy to the style guide. Basically, the style guide captures some "best practices" that reduce bugs. In a similar vein, we could have a "dynamics style guide" that reduces bugs in dynamics-based simulations.

jwnimmer-tri, 2017-02-11:
I see -- I misread the sentence phrasing in the review discussion.

I think "document for readers why you chose a certain literal value" (and therefore have a reason!) is good advice, but I don't see how it's dynamics-specific, nor do I think we are doing badly enough at accomplishing it that we need to spend our precious styleguide words explaining it?

sherm1, 2017-02-11:
I don't think we are ready for this -- for example, I'm not sure I agree with @amcastro-tri's prescription in the comment. (Using a variable step integrator that picks its own step size seems better to me.)

liangfok, 2017-02-11:
Alright, I will close this issue for now, but let's keep it in mind. I suppose any wisdom captured in such a document could be construed to be defects in the dynamics API.

closed 2017-02-11 by liangfok

amcastro-tri, 2017-02-13:
Having a "style guide" for simulation is an interesting suggestion @liangfok. In my experience what works very well in that regard is having a nice set of cases people can refer to. But we are doing that already with the cases we have in "drake/examples". Probably a good thing to keep in mind is to have a nice set of cases for multibody dynamics in the examples directory.
Of course we can , as @sherm1 suggested, use variable time step integrators to pick up the step size for us. But again, I think this is a good discussion to motivate to have a good pool of nice cases with Drake.