RobotLocomotion/drake #5142: RigidBodyTree::centerOfMass() Does Not Ignore Links that are Welded To World

liangfok, 2017-02-11:
# Problem Definition
[RigidBodyTree::centerOfMass()]() incorrectly computes the Center Of Mass of a model instance if the model instance has bodies with non-zero mass that are welded to the world.

File [drake/examples/Pendulum/Pendulum.urdf](https://github.com/RobotLocomotion/drake/blob/daf51977c9f977a44ec703aed5dcd970625eb899/drake/examples/Pendulum/Pendulum.urdf) contains bodies "[world](https://github.com/RobotLocomotion/drake/blob/daf51977c9f977a44ec703aed5dcd970625eb899/drake/examples/Pendulum/Pendulum.urdf#L17)" and "[base_part2](https://github.com/RobotLocomotion/drake/blob/daf51977c9f977a44ec703aed5dcd970625eb899/drake/examples/Pendulum/Pendulum.urdf#L24)" that have non-zero mass but should be ignored since they are welded to the world. The comment in the URDF indicates that they have non-zero mass as a workaround for a bug in `drc-viewer`. Since these bodies are not ignored, their non-zero mass is included in the center-of-mass calculation. This results in an incorrect center of mass of `Z = -0.2425` instead of the expected `Z = -0.5`.

liangfok, 2017-02-11:
@sherm1 and @amcastro-tri, feel free to close this if we're certain such a bug will not be carried over to `MultiBodyTree`.

amcastro-tri, 2017-02-13:
@liangfok, it looks like the code is doing exactly what you are asking. The problem seems to be how to ask for say, the center of mass, for a specific set of your model.
We have model instances, however these are on a per file basis correct? 
how do you think the API to make this kind of request would look like? In this case it seems like the urdf does not provide enough information to tell what is pendulum and what is world. Does SDF allow to specify some sort of grouping or different models per file? can they be connected say with a joint between a body in one model and a second body in another model?

liangfok, 2017-02-13:
> We have model instances, however these are on a per file basis correct?

For URDF, yes since URDFs can only have one model per file. For SDF, no because a single SDF file can have multiple models.

> How do you think the API to make this kind of request would look like?

I think the API should look like `MultiBodyTree::CalcCenterOfMass(int model_instance_id)` where it returns the center of mass of the bodies that belong to the provided `model_instance_id` that are _not_ welded, either directly or transitively, to the world.

> In this case it seems like the urdf does not provide enough information to tell what is pendulum and what is world. 

Actually, the URDF does specify that `base_part_2` is welded to the world by using a `fixed` joint: https://github.com/RobotLocomotion/drake/blob/daf51977c9f977a44ec703aed5dcd970625eb899/drake/examples/Pendulum/Pendulum.urdf#L38. I think the transitive closure of all bodies that are welded to the world should not be included in the center-of-mass calculation. 

> Does SDF allow to specify some sort of grouping or different models per file?

Yes, I believe this is possible through a combination of `model`:

http://sdformat.org/spec?ver=1.6&elem=model

nested `model`:

http://sdformat.org/spec?ver=1.6&elem=model#model_model

and `population` tags:

http://sdformat.org/spec?ver=1.6&elem=world#world_population

I'm not totally confident, however, since I've never used the `population` tag.

> can they be connected say with a joint between a body in one model and a second body in another model?

I suspect this is possible using the nested-`model` tags, but I haven't tried this yet.

jwnimmer-tri, 2018-12-21:
This issue is about the RBT code, in the attic. We're not going to fix it, it's okay to leave it however it is now.

closed 2018-12-21 by jwnimmer-tri