RobotLocomotion/drake #5145: bazel warnings on mac

RussTedrake, 2017-02-12:
fyi — building master w/ bazel on mac, I get spammed with e.g.
```
INFO: From Linking drake/systems/framework/libleaf_system.pic.a:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: bazel-out/osx-opt/bin/drake/systems/framework/_objs/leaf_system/drake/systems/framework/leaf_system.pic.o has no symbols
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: warning for library: bazel-out/osx-opt/bin/drake/systems/framework/libleaf_system.pic.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking drake/systems/framework/libleaf_system.a:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: bazel-out/osx-opt/bin/drake/systems/framework/_objs/leaf_system/drake/systems/framework/leaf_system.o has no symbols
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: warning for library: bazel-out/osx-opt/bin/drake/systems/framework/libleaf_system.a the table of contents is empty (no object file members in the library define global symbols)
```

david-german-tri, 2017-02-13:
Yeah, I've noticed this too, on all our header-only targets.  Thanks for opening an issue.  There's probably a linker flag we can specify in CROSSTOOL to silence it.

david-german-tri, 2017-02-16:
Closing as duplicate of #5183, which has some additional detail.

closed 2017-02-16 by david-german-tri