RobotLocomotion/drake #5148: Set camera options from DrakeVisualizer c++ interface

RussTedrake, 2017-02-13:
@david-german-tri -- while you're thinking about the interactions between drake and drakevisualizer, let me add another feature request.  We should be able to set the default camera angle at the start of a simulation (at least).  This seems necessary for testing our ego car driving amongst a complex traffic scenario; otherwise we'll be finding/clicking the chase cam on every sim.

cc @patmarion

patmarion, 2017-02-13:
btw, I'm not sure how interested you are in viewer-side scripts, maybe not that interested, but I thought I'd just post an example in case it's convenient for anybody.  You can save this code to a script and launch `drake-visualzer --script my_script.py`, and it will initialize the camera tracker anytime an object named 'foo' is added to the scene:

```
def setCameraTracker(obj):
    cameraControlPanel.onSetTarget()
    cameraControlPanel.setTarget(obj)
    cameraControlPanel.ui.trackModeCombo.currentIndex = 3

def onObjectAdded(objectModel, obj):
    if obj.getProperty('Name') == targetName:
        setCameraTracker(obj)

targetName = 'foo'
objectModel.connectObjectAdded(onObjectAdded)
```

caveat- if the object you wish to follow does not have a unique name, the above code is not as useful.  you would have to identify the object by it's full hierarchy name

jwnimmer-tri, 2017-02-13:
That's great.  I was also going to push back on embedding this into the simulation rendering APIs, versus the setup code that configures and starts a simulation (which is in python already).  I think both are valid choices, but the "handle it in the launcher" is likely a cheaper first pass.

david-german-tri, 2017-04-18:
I think the right assignee for this going forward is @SeanCurtis-TRI.

SeanCurtis-TRI, 2017-04-18:
@david-german-tri Acknowledged.

jwnimmer-tri, 2021-11-12:
As I understand it, we're not planning on enhancing drake-visualizer very much anymore.

closed 2021-11-12 by jwnimmer-tri