RobotLocomotion/drake #5151: Buildifier and CLion plugin disagree on deps order

jwnimmer-tri, 2017-02-13:
Given some recent PRs, I suspect that in-tree `tools/buildifier.sh` script and the CLion plugin disagree on `deps=[]` order (for `@otherlibrary//:target` at least, in particular).  They are supposed to agree, so we'll need to find the config or version skew that makes them disagree.

david-german-tri, 2017-04-05:
From inspection of the Bazel CLion plugin's source code, it is running a buildifier binary (ELF) that is simply checked into the source tree as a blob:

https://github.com/bazelbuild/intellij/blob/879aa9f77493ceb35b72e962d3b7d218afbf087c/base/src/com/google/idea/blaze/base/buildmodifier/FileSaveHandler.java

https://github.com/bazelbuild/intellij/blob/master/base/src/com/google/idea/blaze/base/buildmodifier/BuildFileFormatter.java#L40

https://github.com/bazelbuild/intellij/blob/879aa9f77493ceb35b72e962d3b7d218afbf087c/base/src/com/google/idea/blaze/base/buildmodifier/BazelBuildifierBinaryProvider.java

https://github.com/bazelbuild/intellij/blob/master/base/resources/binaries/bazel-buildifier

david-german-tri, 2017-04-05:
-> https://github.com/bazelbuild/intellij/issues/73

david-german-tri, 2017-04-05:
I just tried manually inserting an up-to-date `buildifier` binary into `clwb_bazel.jar`, specifically as `/base/resources/binaries/bazel-buildifer`.  It appears to work perfectly.  I don't think we can recommend that approach to Drake developers at large, though.

It's interesting to note that the plugin definitely isn't able to autoformat BUILD files on OS X, and convenient that it at least fails silently!

david-german-tri, 2017-04-05:
I'm going to close this issue since there's nothing for us to do until https://github.com/bazelbuild/intellij updates their released plugin.  (I'd PR it to them, but they aren't accepting pull requests.)

closed 2017-04-05 by david-german-tri

jwnimmer-tri, 2017-07-27:
I think #6681 may have fixed this for real; the new CLion and Bazel plug-in are reported to get this right.

SeanCurtis-TRI, 2017-07-27:
This does indeed seem to be the case -- although, there's an outside chance that it was fixed when upgrading bazel from 0.4.5 to 0.5.2. There's some anecdotal evidence in this direction. That said, following the current instructions to get CLion to 2017.1.3 seems to correct the issue.

SeanCurtis-TRI, 2017-07-27:
We'll do one more test before closing...

SeanCurtis-TRI, 2017-07-27:
After some more poking, we've come to the following conclusion(s) with CLion 2017.1.3 and 2017.07.05 bazel plugin.

1. It will no longer mess up your files actively.
2. It *may* fix your files (ordering dependencies, etc.)
3. The circumstances under which it fixes your dependencies is somewhat arcane, but seems to be related to adding/deleting carriage returns.

But, from the perspective that it appears to no longer do any harm, I declare this issue closed closed (as opposed to the April "can't-do-anything-about-it" closed.)