RobotLocomotion/drake #5154: Model with zero DOFs welded with kFixed breaks

agalbachicar, 2017-02-13:
I'm trying to load a simple box model through an SDF and visualize it. This simulation has zero degrees of freedom because I just want to load the model with a visualization purpose. 

In case I load the SDF file with `multibody::joints::kFixed` the program ends up with a segmentation fault. This problem is solved by calling `drake::parsers::sdf::AddModelInstancesFromSdfFile` with `multibody::joints::kQuaternion`.

[Here](https://gist.github.com/agalbachicar/1515ca8d64c5fd99b6d7f729d7006836#file-4) you can see the gdb backtrace when using `multibody::joints::kFixed`. And in this [link](https://github.com/ekumenlabs/drake/blob/sdf_sample/drake/sdf_sample/src/sample.cc) you can see a sample project that reproduces the situation.

sherm1, 2017-02-13:
Thanks for reporting this, @agalbachicar and providing such good information. It looks like a problem dealing with the resulting 0-length state vector. We are reworking our multibody code at the moment so probably won't fix this in the old code but will make sure it works in the new code. We would also be very glad to accept a pull request that fixes the current code!

@amcastro-tri, assigning to you for tracking in MultibodyTree.

jwnimmer-tri, 2017-02-13:
It will be some common, when using PoseAggregator in automotive, for the fixed scene (world geometry) to come in via SDF (RBT) and all of the moving joints to come in as Systems that don't use RigidBodyPlant.  If and when that happens, I will bump the priority of this accordingly.

jwnimmer-tri, 2019-06-20:
Closing as "won't fix" because this is about the old RigidBodyTree APIs.

closed 2019-06-20 by jwnimmer-tri