RobotLocomotion/drake #5162: (x * (y + z)).EqualTo(x*y + x*z) returns false.

hongkai-dai, 2017-02-14:
In the current symbolic expression, it does not handle the parenthesis well. The comparison
```C++
(x * (y + z)).EqualTo(x*y + x*z)
```
returns false, when comparing the hash values in https://github.com/RobotLocomotion/drake/blob/master/drake/common/symbolic_expression.cc#L120. But even if I commented out the hash value comparison, it still returns false.

soonho-tri, 2017-02-14:
This is a duplicate of https://github.com/RobotLocomotion/drake/issues/5087. `Expression::EqualTo` checks structural equality. If you do not expand the two terms, they are structurally different.

soonho-tri, 2017-02-14:
As we discussed in f2f: I'll add extra documentation clarifying `Expression::EqualTo` method.

PR #5244 (2017-02-21): Expand comment section in symbolic::Expression::EqualTo
Close #5162

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5244)
commit 1efc4c98 (2017-02-21): Expand comment section in symbolic::Expression::EqualTo (#5244)

* Expand comment section in symbolic::Expression::EqualTo

Close #5162
closed 2017-02-21 by hongkai-dai