RobotLocomotion/drake #5172: Add derivatives to PiecewisePolynomialTrajectory

Lucy-tri, 2017-02-14:
We would like to have the facility to create PiecewisePolynomialTrajectory’s with their associated derivatives, up to a specified derivative order.  Siyuan implemented something similar, a PiecewisePolynomialSource which is a System, in https://reviewable.io/reviews/RobotLocomotion/drake/4992#.  Instead of that design, we would like PiecewisePolynomialTrajectory to be associated with it’s derivatives. 

The PiecewisePolynomialTrajectory class’s constructor can be modified to have an optional parameter, derivative_order (an int). This will be used to create the derivatives of the PiecewisePolynomialTrajectory, up to the specified derivative order.

Should the functionality should be added to the Trajectory class instead?
Currently, the only subclass of Trajectory is PiecewisePolynomialTrajectory, so we will implement the needed functionality in PiecewisePolynomialTrajectory class now. If ever another type of Trajectory is created, we can then add the ability to specify derivative order there, instead of in PiecewisePolynomialTrajectory.

See the class PiecewisePolynomialSource in kuka_inverse_dynamics_demo.cc (PR #4992) for implementation of derivative order.

Lucy-tri, 2017-02-14:
@sammy-tri @siyuanfeng-tri @RussTedrake : you may be interested.

RussTedrake, 2017-02-15:
I'm a little confused.  Taking the derivatives of a piecewise polynomial is trivial (you can simply read out from the existing coefficient matrix slightly differently).  I don't think there is any need to specify a derivative order in the constructor.  

I think that the two most natural interfaces would be:
1) add a method similar to `value(double t)` that is something like `derivative(double t, int order)`
2) add a method that extracts an entire trajectory of the derivatives... e.g. `derivativeTrajectory(int order)` that creates a new `PiecewisePolynomialTrajectory`.

Both would be reasonable at the `Trajectory` level, if we have another trajectory class that also implements these.

siyuanfeng-tri, 2017-02-15:
The longer version of the story is that I wanted a vector of q, qd, qdd as input, and in the current System primitives, we only had a TrajectorySource that spits out q. TrajectorySource is constructed from a Trajectory, and because of the abstract interface of Trajectory, which doesn't have derivatives, I can't compute the derivatives. So I made primitive similar to TrajectorySource, but constructed from PP directly, and it's output is q, qd, qdd, ... 

I agree the most suitable thing to is to have a derivative API in Trajectory.

sammy-tri, 2017-02-15:
So if we add a derivative() call to Trajectory, does TrajectorySource gain a derivative_output_port() (one for each derivative order)?

GetDerivativeTrajectory() might make more sense because then you can just make more TrajectorySource systems.

I'm still not opposed to just making the output of value() wider (with the resulting change on the size of the TrajectorySource output port) either.

siyuanfeng-tri, 2017-02-15:
the most intuitive thing for me is to have a derivative call in Trajectory, and have TrajectorySource output a single vector output that outputs the derivatives as well.

jwnimmer-tri, 2017-02-15:
You could also imagine that `TrajectorySource` offers (via constructor or factory method) a way to choose the nature of the output port -- whether its just the trajectory, or also stacks up some derivatives into the same output port.

sammy-tri, 2017-02-15:
> the most intuitive thing for me is to have a derivative call in Trajectory, and have TrajectorySource output a single vector output that outputs the derivatives as well.

I guess we can cross the bridge of a Trajectory subclass which is unable to implement derivative() when we come to it (allow derivative() to throw or something).

Lucy-tri, 2017-02-17:
I'm working on this issue.

Lucy-tri, 2017-02-28:
Part 2 will be to update TrajectorySource with a constructor that can specify a derivative order (the number of derivatives to output).

Lucy-tri, 2017-03-21:
Done (both part 1 and part2).

closed 2017-03-21 by Lucy-tri