RobotLocomotion/drake #5179: Generalized LCM message playback (with focus on visualization)

SeanCurtis-TRI, 2017-02-15:
# Problem

PR #5152 gives `DrakeVisualizer` the ability to cache the relevant time-stamped `RigidBodyPlant` state and then play it back at real time.  It is a convenient utility but has limited scope.  For example, this approach would be unable to play back and simulation artifacts which are published in different systems (e.g., contact forces which are handled by a different `System`.)  We can make a more general and powerful solutions.

# Proposed Solution

The `DrakeVisualizer` system owns a pointer to a `drake::lcm::DrakeLcmInterface` instance.  It is responsible for *constructing* LCM messages, but passes them to the LCM interface for broadcasting.  This interface provides a method for injecting a more general caching and playback solution.

We provide a new implementation of the `drake::lcm::DrakeLcmInterface` (perhaps `PlaybackDrakeLcm`).  It contains a pointer to the current implementation and caching/playback functionality.  `System`s that currently take a pointer to a `DrakeLcmInterface` would be use an instance of `PlaybackDrakeLcm` in place of `DrakeLcm`.  The logic added by #5152 to `DrakeVisualizer` would then be implemented in `PlaybackDrakeLcm` with the following details:
   
   - Rather than caching (timestamp, `Eigen::Vector`) pairs, it would cache (timestamp, bytestring) pairs.
   - `PlaybackDrakeLcm` would *not* be able to use the `PiecewisePolynomial`; the "value" data type is incompatible (bytestring vs `Eigen::vector`).  Instead, it would have to implement its own zero-order hold to select the correct message for an arbitrary time value.  At this point, it doesn't seem that higher-order interpolation is actually necessary.
   - `PlaybackDrakeLcm` would contain a *whitelist* of LCM message types to cache.

## Possible issues

- Generally, LCM messages are probably bigger (in memory footprint) then the underlying data that produced them and would lead to redundancies (e.g., the `DrakeVisualizer` only caches `RigidBodyPlant` state, even though the full message includes body names; it relies on structural consistency to map the two.)  The size of the messages should be evaluated to see if that is a problem with possible mitigating strategies:
   - Compress the messages
   - Cache values at a different rate than messages are published (particularly if the playback rate is significantly lower than the simulation rate).

jwnimmer-tri, 2017-02-15:
Your ideas are intriguing to me and I would like to subscribe to your newsletter!

(In other words, :+1:.)

liangfok, 2017-02-15:
The proposed design sounds good and is certainly more general than the current solution.
The inability to interpolate may become a concern if the we want to do _ultra slow motion replay_.

BTW, there is an extra "be" in "would be use".

liangfok, 2017-02-15:
Another design idea is to add support for listeners to `lcm::DrakeLcmInterface`. Something like `lcm::DrakeLcmInterface::AddListener()`. One listener would be to log and replay message bytes.

SeanCurtis-TRI, 2017-02-15:
1) Typo fixed (thanks @liangfok)
2) Interpolation is a bit problematic; you'd have to have an interpolator that a) knows the structure of the message and b) knows the nature of the integration such that meaningful numerical interpolation can be performed.

liangfok, 2017-02-15:
Yes, I agree interpolation will be a challenge. I can't think of a good design at the moment that would enable interpolation. Maybe we'll just drop support for that until it really becomes a problem.

patmarion, 2017-02-18:
Just to state the obvious... LCM already has logging and playback tools.  I know you guys are discussing a custom implementation inside Drake, but have you already ruled out using LCM's provided tools?  They work quite well and have many useful features.  If they don't solve your use case exactly, perhaps consider improving the LCM tools?

jwnimmer-tri, 2017-02-18:
Agreed.  I continue to believe that `lcm-logger` to capture plus inline-timestamp-awareness in `lcm-logplayer` (or a python log-conversion tool that has the same effect) would be the best answer.

Heck, even Director (or an associated python tool in the same family) could offer to open an LCM logfile and scrub through the visualizer-related messages for playback in realtime or locked sub/super-realtime.

I don't understand the need to do it in-memory like this issue (or the PR that inspired it) suggests, but I think if we _are_ going to do it in-memory, then this issue's approach is better than what went into master.

patmarion, 2017-02-18:
If Director tools are a consideration, I have some lcm log utilities that are relevant.  I just made this video to demonstrate:

https://www.youtube.com/watch?v=oK5hHMIp_Ec

I used existing code in Director and wrapped a simple gui around it for demonstration.  It's a programmatic log playback interface that I developed a while ago to automate movie production.  The tool can playback any lcm log, clocked by the original log's wall time, or clocked using a timestamp field from the lcm messages.

SeanCurtis-TRI, 2017-02-21:
@patmarion Can you clarify what you mean by "clocked by"?  There are two orthogonal timings involved here: the timestamp of the data and the playback rate.  In this case, it seems we have two timestamps on the data: the computer time when a message was received and the simulation time stamp that was stashed in the data (recognizing that we have no guarantees that the former is an affine transformation of the latter.)  But what about playback?  One of the important features is to playback in realtime based on the simulation time stamps in the data.  How close is that?

patmarion, 2017-02-21:
> playback in realtime based on the simulation time stamps in the data

That's what my video is showing.  The tool works very well and I use it for making movies.  The code looks like this:

```
def getDrakeSimTimeForEvent(event):
    '''For compatible events, returns time in microseconds for the logged event'''
    if event.channel.startswith('DRAKE_VIEWER'):
        msg = lcmspy.decodeMessage(event.data)
        if hasattr(msg, 'timestamp'):
           return msg.timestamp*1000

filename = 'sim_funnel_grasp.lcmlog'
print 'reading log:', filename

logPlayer = LcmLogPlayer()
logPlayer.readLog(filename, eventTimeFunction=getDrakeSimTimeForEvent)
```

Logs may also contain messages without usable time information.  For example, the message to load geoemetry, which is critical for log playback, does not have a timestamp!

https://github.com/RobotLocomotion/drake/blob/master/drake/lcmtypes/lcmt_viewer_load_robot.lcm

jwnimmer-tri, 2019-11-01:
@SeanCurtis-TRI This is 2+ year old issue still relevant?  Are we planning on doing fancy within-Diagram things for message playback?  Empirically, it seems like "no" is the answer.  We are using lcm-logger and lcm-logplayer extensively already.

SeanCurtis-TRI, 2019-11-01:
Being able to do looping within a single process seems convenient. But this is so far off anyone's radar. That we can close it with the caveat that if it miraculously becomes relevant in the future, we can reopen it.

closed 2019-11-01 by jwnimmer-tri