RobotLocomotion/drake #5180: Bazel error with `*`

soonho-tri, 2017-02-15:
Repro step: `bazel build //drake/common:*`

Errors:
```
INFO: Found 279 targets...
ERROR: missing input file '//drake/common:cpplint.py'.
ERROR: /home/soonhok/work/drake-distro/drake/common/BUILD:516:1: //drake/common:cpplint.py: missing input file '//drake/common:cpplint.py'.
ERROR: /home/soonhok/work/drake-distro/drake/common/BUILD:516:1 1 input file(s) do not exist.
INFO: Elapsed time: 0.205s, Critical Path: 0.00s
```

/cc @david-german-tri , @jwnimmer-tri

soonho-tri, 2017-02-15:
@david-german-tri suspects that it comes from https://github.com/RobotLocomotion/drake/blob/master/tools/cpplint.bzl#L55

Reference: https://bazel.build/versions/master/docs/bazel-user-manual.html

david-german-tri, 2017-02-15:
> @david-german-tri suspects that it comes from https://github.com/RobotLocomotion/drake/blob/master/tools/cpplint.bzl#L55

That's now confirmed.

PR #5181 (2017-02-16): Specify lint tools as absolute labels.
This better conforms with the py_test documentation, and prevents `//drake/foo:all-targets` from treating `cpplint.py` as a package-local file, which it isn't.

https://bazel.build/versions/master/docs/be/python.html#py_test

Fixes #5180.

@soonho-tri feature, @jwnimmer-tri platform

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5181)
commit 028c726e (2017-02-16): Specify lint tools as absolute labels.

This better conforms with the py_test documentation, and prevents `//drake/foo:all-targets` from treating cpplint.py as a package-local file, which it isn't.

https://bazel.build/versions/master/docs/be/python.html#py_test

Fixes #5180.
closed 2017-02-16 by jwnimmer-tri