RobotLocomotion/drake #5183: Numerous "clang: warning: argument unused during compilation: '-pthread'" Warnings When Building With Bazel on OS X

liangfok, 2017-02-15:
# Sample Output

```
INFO: From Linking drake/common/cond_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/common/trig_poly_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/common/double_overloads_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/common/trajectories/piecewise_polynomial_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/common/trajectories/exponential_plus_piecewise_polynomial_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/common/trajectories/piecewise_generation_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/math/autodiff_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/common/polynomial_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/math/cross_product_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/math/gradient_util_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/math/expmap_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/math/normalize_vector_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/common/autodiff_overloads_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/common/trajectories/qp_spline/spline_generation_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/math/rotation_conversion_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/math/jacobian_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/common/trajectories/piecewise_quaternion_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/multibody/collision/model_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/multibody/joints/joint_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/common/trajectories/piecewise_polynomial_trajectory_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/multibody/collision/fcl_model_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Compiling drake/automotive/automotive_simulator.cc:
drake/automotive/automotive_simulator.cc:337:20: warning: 'AddModelInstanceFromUrdfFile' is deprecated:
DRAKE DEPRECATED: Please use AddModelInstanceFromUrdfFileWithRpyJointToWorld(). [-Wdeprecated-declarations]
    parsers::urdf::AddModelInstanceFromUrdfFile(urdf_filepath,
                   ^
drake/automotive/automotive_simulator.cc:375:16: note: in instantiation of member function 'drake::automotive::AutomotiveSimulator<double>::Start' requested here
template class AutomotiveSimulator<double>;
               ^
./drake/multibody/parsers/urdf_parser.h:242:22: note: 'AddModelInstanceFromUrdfFile' has been explicitly marked deprecated here
ModelInstanceIdTable AddModelInstanceFromUrdfFile(
                     ^
1 warning generated.
INFO: From Linking drake/common/libcond.pic.a:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: bazel-out/osx-opt/bin/drake/common/_objs/cond/drake/common/cond.pic.o has no symbols
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: warning for library: bazel-out/osx-opt/bin/drake/common/libcond.pic.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking drake/systems/framework/value_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/systems/framework/system_output_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/math/matrix_util_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/systems/framework/supervector_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/systems/framework/system_input_test:
clang: warning: argument unused during compilation: '-pthread'
[... 190 lines cut ...]
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/multibody/parsers/parser_common_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/examples/QPInverseDynamicsForHumanoids/lcm_utils_test:
clang: warning: argument unused during compilation: '-pthread'
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/multibody/benchmark_rigid_body_tree:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/examples/QPInverseDynamicsForHumanoids/param_parsers/param_parser_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/automotive/automotive_simulator_test:
clang: warning: argument unused during compilation: '-pthread'
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/multibody/debug_manipulator_dynamics:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/multibody/parsers/urdf_manipulator_dynamics_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/multibody/parsers/urdf_kinematics_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/multibody/parsers/urdf_collision_test:
clang: warning: argument unused during compilation: '-pthread'
INFO: From Linking drake/automotive/automotive_demo:
clang: warning: argument unused during compilation: '-pthread'
clang: warning: argument unused during compilation: '-pthread'
```

liangfok, 2017-02-15:
@david-german-tri please feel free to close if we're not planning on supporting Bazel + OSX.

jwnimmer-tri, 2017-02-15:
Yes, we are supporting Bazel + OSX.  Fixing warnings is not super high on the list though; making it support all Drake features comes first.

liangfok, 2017-02-15:
OK. There are a couple other types of warnings that flood my terminal when I build using Bazel on OSX. Here are samples:

```
ld: warning: could not create compact unwind for ___dmumps_parallel_analysis_MOD_dmumps_715: stack subq instruction is too different from dwarf stack size
ld: warning: could not create compact unwind for _write_float: stack subq instruction is too different from dwarf stack size
```

And:

```
INFO: From Linking drake/common/libcond.a:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: bazel-out/osx-opt/bin/drake/common/_objs/cond/drake/common/cond.o has no symbols
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: warning for library: bazel-out/osx-opt/bin/drake/common/libcond.a the table of contents is empty (no object file members in the library define global symbols)
```

jwnimmer-tri, 2017-02-15:
The final item is a dup of #5145.

liangfok, 2017-02-15:
Is there a plan for adding OSX + Bazel to either [Continuous Production](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/) or [Nightly Production](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/)?

Or is it already there and I simply missed it? (I was trying to verify that the same warnings show up in CI).

jwnimmer-tri, 2017-02-15:
Other than PRs it's only in https://drake-jenkins.csail.mit.edu/view/Bazel/job/mac-clang-bazel-nightly-release/ per https://drake-jenkins.csail.mit.edu/view/Bazel/.  Yes, we are working on making it be production-supported in Continuous.

liangfok, 2017-02-15:
Thanks. I just confirmed that all aforementioned warnings show up in CI, which is good.

liangfok, 2017-04-01:
#5662 resulted in the following warning appearing:
```
INFO: From Linking drake/automotive/liblane_direction.a:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: bazel-out/osx-opt/bin/drake/automotive/_objs/lane_direction/drake/automotive/lane_direction.o has no symbols
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: warning for library: bazel-out/osx-opt/bin/drake/automotive/liblane_direction.a the table of contents is empty (no object file members in the library define global symbols)
```

david-german-tri, 2017-04-14:
Reassigning to @soonho-tri.

jamiesnape, 2018-08-02:
Just noticed:
```
DEBUG: pkg_config.bzl:186:9: pkg-config of protobuf returned flags that we will ignore: ["-Qunused-arguments"]
```
Hence, targets that use `protobuf` have the following:
```
clang: warning: argument unused during compilation: '-pthread'
```

Since I missed that for weeks, I would advocate having the `DEBUG` message from `pkg-config.bzl`having the text `WARNING` in it so that CDash highlights and having a way to declare flags (such as `-Qthread`) that we know we should ignore so that the warning can be suppressed.

Also, having a platform dependent way declare flags that should be ignored could strip out the bad `-pthread` on Mac.

EricCousineau-TRI, 2018-12-13:
Per f2f, still waiting for `bazelbuild`s upstream update. Any other warnings can be filtered out via CDash.

EricCousineau-TRI, 2018-12-13:
Per f2f, closing this generic issue as we have the xref'd specific issues that provide better insight.

closed 2018-12-13 by EricCousineau-TRI

jamiesnape, 2019-01-10:
For the record, also related is bazelbuild/bazel#5981, but I do not think it deserves a Drake issue or this issue being reopened.