RobotLocomotion/drake #5186: Mysterious CI Failure: "Workspace has a .git repository, but it appears to be corrupt"

liangfok, 2017-02-16:
# The Error

```
22:10:25 ERROR: Workspace has a .git repository, but it appears to be corrupt.
22:10:25 hudson.plugins.git.GitException: Command "git rev-parse --is-inside-work-tree" returned status code 128:
22:10:25 stdout: 
22:10:25 stderr: fatal: Not a git repository (or any of the parent directories): .git
22:10:25 
22:10:25 	at org.jenkinsci.plugins.gitclient.CliGitAPIImpl.launchCommandIn(CliGitAPIImpl.java:1784)
22:10:25 	at org.jenkinsci.plugins.gitclient.CliGitAPIImpl.launchCommandIn(CliGitAPIImpl.java:1757)
22:10:25 	at org.jenkinsci.plugins.gitclient.CliGitAPIImpl.launchCommandIn(CliGitAPIImpl.java:1753)
22:10:25 	at org.jenkinsci.plugins.gitclient.CliGitAPIImpl.launchCommand(CliGitAPIImpl.java:1409)
22:10:25 	at org.jenkinsci.plugins.gitclient.CliGitAPIImpl.launchCommand(CliGitAPIImpl.java:1421)
22:10:25 	at org.jenkinsci.plugins.gitclient.CliGitAPIImpl.hasGitRepo(CliGitAPIImpl.java:196)
22:10:25 	at hudson.plugins.git.GitAPI.hasGitRepo(GitAPI.java:232)
22:10:25 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
22:10:25 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
22:10:25 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
22:10:25 	at java.lang.reflect.Method.invoke(Method.java:498)
22:10:25 	at hudson.remoting.RemoteInvocationHandler$RPCRequest.perform(RemoteInvocationHandler.java:894)
22:10:25 	at hudson.remoting.RemoteInvocationHandler$RPCRequest.call(RemoteInvocationHandler.java:869)
22:10:25 	at hudson.remoting.RemoteInvocationHandler$RPCRequest.call(RemoteInvocationHandler.java:828)
22:10:25 	at hudson.remoting.UserRequest.perform(UserRequest.java:153)
22:10:25 	at hudson.remoting.UserRequest.perform(UserRequest.java:50)
22:10:25 	at hudson.remoting.Request$2.run(Request.java:336)
22:10:25 	at hudson.remoting.InterceptingExecutorService$1.call(InterceptingExecutorService.java:68)
22:10:25 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
22:10:25 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
22:10:25 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
22:10:25 	at java.lang.Thread.run(Thread.java:745)
```

# Example Logs
* https://drake-jenkins.csail.mit.edu/job/linux-gcc-experimental-ros/3717/

david-german-tri, 2017-02-16:
Network is known to be a little flaky, and git is maybe a little flaky also.  I'm prioritizing this as `backlog` for now.

jwnimmer-tri, 2017-03-11:
We can track here, and then repoen when we see too many.

closed 2017-03-11 by jwnimmer-tri